wled/WLED #2526: Option to disable CCT channel because it is not supported by HA integration anymore

Pro75357, 2022-02-05:
**Is your feature request related to a problem? Please describe.**
Home Assistant integration no longer allows CCT channel as of 2022.2.2 For some reason this was not mentioned in the breaking changes, but it breaks integrations that use WLED with a CCT channel. It did not support CCT channel gracefully (colors were inaccurate), so now it seems it is not allowed. 

**Describe the solution you'd like**
A toggle switch in the LED configuration settings to disable CCT channel, or do not report a CCT channel at all when "calculate CCT from RGB is enabled" (it would also be nice if the HA integration more gracefully handled this situation, but the issue was closed). 

**Describe alternatives you've considered**
I tried to enable "Calculate CCT from RGB", but this apparently does not disable the CCT channel from being reported. I tried all combination of settings without success.

**Additional context**
Please see https://github.com/home-assistant/core/issues/65616

Thank you for your ideas for making WLED better!
WLED is awesome :)

Aircoookie, 2022-02-05:
Hey, thank you for the report!
In addition to enabling "Calculate CCT from RGB", also make sure "White Balance correction" is disabled.
If there is no longer a color temperature slider in the main WLED UI, HA should accept the device as well :)

Aircoookie, 2022-02-06:
Let me add a quick "wontfix" statement here as it is not in my interest to support evading the detection as a CCT device by the HA integration.
While I don't agree with Frenck's decision to block these devices, I still respect it, and #2500 is in the works to resolve this problem.

closed 2022-02-06 by Aircoookie

Aircoookie, 2022-02-06:
@Pro75357 If the device is still rejected even though  "Calculate CCT from RGB" and "White Balance correction" is disabled, please let me know. In that case there might a) be a bug in WLED that causes it to report itself as CCT even though it isn't, or b) a bug in the integration that flags a device as CCT even though it isn't.

Pro75357, 2022-02-07:
@Aircoookie   - disabling "White balance correction" seems to have done the trick. Thank you for your help!

jdavis7765, 2022-04-24:
@Aircoookie I see you closed [this issue](https://github.com/Aircoookie/WLED/issues/2500) out you linked to here. Does that mean it should be supported in HA or that the HA side may still need to be implemented by them? 

I just got some Athom Bulbs and couldn't bring them into HA without checking "Calculate CCT from RGB" but I lose control of just the white channel this way. 

Thanks!

blazoncek, 2022-04-24:
You may need to contact HA developers.
WLED implemented a new way to inform integrations about light capabilities it supports. HA may still need to implement those.