wled/WLED #2535: Macro RESET_RUNTIME uses memset() on non-trivial type

henrygab, 2022-02-08:
This compiler warning appears legitimate.  What is the macro really attempting to do here?

<details><summary>Compiler puts out the following warning</summary><P/>

```
In file included from wled00\fcn_declare.h:132,
                 from wled00\wled.h:145,
                 from wled00\FX_fcn.cpp:26:
wled00\FX_fcn.cpp: In member function 'void WS2812FX::finalizeInit()':
wled00\FX.h:84:80: warning: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'WS2812FX::segment_runtime' {aka 'class WS2812FX::Segment_runtime'}; use assignment or value-initialization instead [-Wclass-memaccess]
   84 | #define RESET_RUNTIME    memset(_segment_runtimes, 0, sizeof(_segment_runtimes))
      |                                                                                ^
wled00\FX_fcn.cpp:70:3: note: in expansion of macro 'RESET_RUNTIME'
Retrieved `.pio\build\esp8266_2m\libb18\IRremoteESP8266\ir_Lutron.cpp.o' from cache
   70 |   RESET_RUNTIME;
      |   ^~~~~~~~~~~~~
In file included from wled00\fcn_declare.h:132,
                 from wled00\wled.h:145,
                 from wled00\FX_fcn.cpp:26:
wled00\FX.h:362:20: note: 'WS2812FX::segment_runtime' {aka 'class WS2812FX::Segment_runtime'} declared here
  362 |     typedef struct Segment_runtime { // 28 bytes
      |                    ^~~~~~~~~~~~~~~
```

</details>

The compiler appears to be correctly warning about a bad practice.

Put simply, the `Segment_runtime` structure contains an embedded pointer ... namely member `byte * data`.  Therefore, it should never be blindly `memset` to zero.

It appears that this will cause a memory leak, if there is data currently allocated (as evidenced by a non-`nullptr` value in member `data`).

Note: I got this error building default boards using platform io.
(I believe I saw this error on esp8266_2m)

This macro is only used in one location: FX_fnc.cpp at `WS2812FX::finalizeInit(void)`.   I'm not entirely sure what the "correct" fix is.  Perhaps this was just missed when fixing earlier multi-core issues with resetting of `_segment_runtimes`?  That said, here's a strawman to start discussion...

Remove the macro definition, and replace it's one use with:
```c++
  for (uint16_t i = 1; i < MAX_NUM_SEGMENTS; i++)
  {
    _segment_runtimes[i].reset();
  }
  _segment_runtimes[0].reset();
```

It's also odd ... two places that reset this loop from 1..N, then reset index zero, while a third just loops through them all.  This suggests that there may be latent bugs, such as by relying on an order of writes that's not actually guaranteed by the compiler?

blazoncek, 2022-02-08:
You may have a point. If user would be changing buses (led configuration) often during runtime. In such case memory pointed by `data` member of _segment_runtimes[i] will not be freed. But luckily this does not happen often.
The solution would be to add `for (i=0; i<MAX_NUM_SEGMENTS; i++) if (_segment_runtimes[i].data) _segment_runtimes[i].deallocateData();` in wled.cpp before initializing buses (i.e. checkSegmentAlignment() would be a good candidate) or after in finalizeInit() (in front of of RESET_RUNTIME).

Aircoookie, 2022-02-08:
Thank you for pointing this out, it absolutely causes a memory leak when re-initing busses while data is allocated.

While I'm at it, I will change the name of `segment_runtime::reset()` as this does not actually reset the segment runtime but only marks it for reset (reason is that on ESP32 you can't reset it on core 1 if the effect function is executing on core 0, or you'll get a null pointer exception)

PR #2536 (2022-02-09): Fix re-init segment data leak
Fixes #2535
commit 00b0193a (2022-02-09): Fix re-init segment data leak (fixes #2535 ) (#2536)
closed 2022-02-09 by Aircoookie