wled/WLED #2537: esp32s2_saola warnings: <command-line>: warning: ISO C++11 requires whitespace after the macro name

henrygab, 2022-02-09:
This appears quite often for the first build, and will appear for each source file compiled.

This warning appears when a define passed via command-line parameter contains illegal characters in the identifier.  If you build verbose (under advanced group for the board), the full command line for the compiler is output.  The command line has the following invalid option:

`-DARDUINO_PARTITION_tools/WLED_ESP32_4MB_1MB_FS`

The partition choice is set via `board_build.partitions` in a given environment, and the value appears to be correct.  Not that it *should* change anything, but changing from:
`default_partitions = tools/WLED_ESP32_4MB_1MB_FS.csv`
to:
`board_build.partitions = ${esp32.default_partitions}`
did not change the result.

Can anyone else pickup the trail from here?  Is this a bug in the underlying ESP32s2 arduino support, or is it something in WLED's configuration?

Aircoookie, 2022-02-11:
Thank you for the research :)

It is likely that this is a bug in the general PlatformIO arduino support, or in the [espressif32 platform](https://github.com/platformio/platform-espressif32)

Specifically, the presence of the character `/` invalidates this macro, which leads me to believe that the code that converts `board_build.partitions` to the compile flag has no concept of directories (and potentially characters legal in file names but not C macros)

This problem is by the way not limited to esp32s2, but to all ESP32 builds using the newer ESP32 Arduino core 2.0.x (WLED branch arduinocore2, adoption for regular ESP32 is frozen currently because of #2518). It is not yet officially supported by the espressif32 platformio platform, but required for S2 and C3 support.

henrygab, 2022-02-11:
Yes, the '/' character is illegal for the identifier, although it would be legal as the _content_ of an identifier (such as `-DARDUINO_PARTITION=...`).

Is this **_intentionally_** part of the identifier, then?

I had thought the problem was two part:

(a) incorrect name of the identifier (e.g., should be
     `-DARDUINO_PARTITION=tools/WLED_ESP32_4MB_1MB_FS`)
(b) failing to wrap in quotes (e.g., and then more correctly:
     `-DARDUINO_PARTITION="tools/WLED_ESP32_4MB_1MB_FS"`)

I suppose I didn't dig far enough into how this define is used in pre-core 2.x.

henrygab, 2022-07-13:
foo

softhack007, 2023-01-06:
Superseded by recent activities on 0.14.x to support new MCUs like ESP32-S2 (still experimental). DARDUINO_PARTITION is not used in platformio.ini any more.

closed 2023-01-06 by softhack007