wled/WLED #2542: APA102 don´t work with LAN8720 (WT32-ETH01)

AVRnet, 2022-02-12:
### What happened?

With actual build 2112080 i have a problem to bring WT32-ETH01 LAN Interface work together with APA102 LEDs.

### To Reproduce Bug

Configured LAN Interface (W32-ETH01) and selection of APA102 with save, the Modul run into bootloop and AccessPoint go up.

### Expected Behavior

LAN Interface is configured and works amazing with WS281x and SK6812 LEDs.

### Install Method

Binary from WLED.me

### What version of WLED?

2112080

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-02-12:
Use GPIO pins that do not conflict.

AVRnet, 2022-02-15:
I have tested all kinds of GPIO combinations.
Only WLAN works flawless with APA102 Chips.
I couldn´t have done with W32-ETH01 and APA102 to use the LAN port.

Setup with WS281x and SK6812 LEDs works however perfect.

blazoncek, 2022-02-25:
Try lowering SPI speed by changing DotStarSpiHzMethod; in bus_wrapper.h

lechk82, 2022-03-21:
I'm having the same issue. lowering SPI speed does not help. Did you solve the problem?

AVRnet, 2022-03-22:
I'm afraid not. I haven´t got any idea to solve the problem.

lucasheld, 2022-05-16:
I also have the same problem.
I am using WS2801 LED strips with an ESP32 dev board and a LAN8720 module. The GPIO pins were connected as on a WT32-EHT01, so WLED works without adjustment with the provided binary.
All LED strips that require only one pin can be selected without any problems. As soon as a LED strip is selected that needs clock and data pins, Ethernet does not work anymore and the access point is started. Only after selecting another LED strip Ethernet works again.
No possible GPOI combination works. Also reducing the SPI speed does not change anything.
To reproduce the behavior the LED strip does not have to be present and connected, the selection in the web interface is sufficient.

blazoncek, 2022-05-16:
SPI is used exclusively for driving APA102 (or other SPI) LEDs.
No other SPI device can currently coexist with SPI LEDs.

lucasheld, 2022-05-17:
As far as I know, the LAN8720 module does not use SPI at all, but RMII. Accordingly SPI should be available for the LEDs right?

d3473r, 2022-06-29:
I bought a WT32-EHT01 to run a WS2801 strip and I'm now facing the same issue, ethernet not working as soon as WS2801 is selected in WLED.

The datasheet mentions RMII not SPI as @lucasheld mentioned.
Maybe someone who speakes chinese can help with the datasheet?
[WT32-ETH01规格书V1.2.pdf](https://github.com/Aircoookie/WLED/files/9008937/WT32-ETH01.V1.2.pdf)

lucasheld, 2022-06-30:
@d3473r
The datasheet is also available in english.
[WT32-ETH01_datasheet_V1.1- en.pdf](https://github.com/Aircoookie/WLED/files/9017709/WT32-ETH01_datasheet_V1.1-.en.pdf)

ywwg, 2022-08-22:
[image]

ywwg, 2022-09-17:
Can confirm, as soon as I set LED type to APA102 ethernet stops working.  Also I am using software SPI, because I can change the GPIO pins at will

BlackBird77, 2022-09-30:
Hi All.. 

i can confirm it too.. 
If you use SPI LEDs and Ethernet it is not working.. 

Here are the modes:
DotStarMethod: **OK** but Software only
DotStarEsp32DmaVspiMethod:  **OK** but to fast
DotStarEsp32DmaVspi1MhzMethod:  **OK** perfect
DotStarSpiMethod: NotOK should work, but it did not...
DotStarVspiMethod: NotOK should work, but it did not...
DotStarEsp32DmaHspiMethod: NotOK because it conflicts with ethernet port
DotStarHspiMethod: NotOK because it conflicts with ethernet port

Happy lights
André

blazoncek, 2022-09-30:
WLED uses HSPI for driving LEDs as VSPI may be used by other devices.

BlackBird77, 2022-10-07:
> WLED uses HSPI for driving LEDs as VSPI may be used by other devices.

On the Olimex Board is the ethernet controller on Hspi and the SD Card on Vspi.. If you don't use a external SD card you can drive the Leds on Vspi..

And if i read the code right, the Arduino default is also Vspi, but it initializes only some spi IO pins and not the QuadHD, QuadWP Pins of Vspi.

The Esp32 DMA SPI mode initializes all Vspi pins and deactivate the unused pins..

zigomatichub, 2022-10-26:
I'm having issue with WT32-ETH01 too but not using the APA ledstrip. 

It's only work at the configuration time. I have compile some versions and doing a lot of testing and still searching how to solve the issue I'm facing.

Here the output at configuration time : (from wifi setup - experimental -  ethernet : none to wt32-eth01 -> save)
[image]

1. WLED without Lan configured starting in Wifi (cable removed as well , if not it's seems logicaly different)
lastReconnectAttempt == 0
Access point disabled (init).
Connecting to wifi...
Connected! IP address: 10.0.4.103

2. I put the cable -> no led - no connectivity (as foreseen)
3. WLED: change config and activate the LAN and save

initE: Attempting ETH config: 1
initC: *** Ethernet successfully configured! ***
ETH StartedWriting settings to /wsec.json...
Writing settings to /cfg.json...
Forcing reconnect.
WS client disconnected.
Access point disabled (init).
Connecting to wifi...
Connected! IP address: 10.0.4.103
Init STA interfaces
mDNS started
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
ETH ConnectedDisconnected!
Access point disabled (init).
Connecting to wifi...
Last reconnect too old.
Access point disabled (init).
Connecting to wifi...

Connected! IP address: 10.0.4.109
--> The WLED is now connected in LAN.

4. If I power OFF, ON:
initC: *** Ethernet successfully configured! ***
Removing all.
lastReconnectAttempt == 0
Access point disabled (init).
Connecting to wifi...

Connected! IP address: 10.0.4.103
Init STA interfaces
mDNS started
ETH ConnectedDisconnected!
Access point disabled (init).
Connecting to wifi...
Last reconnect too old.
Access point disabled (init).
Connecting to wifi...
Last reconnect too old.
Access point disabled (init).
Connecting to wifi...

Connected! IP address: 10.0.4.103

I can't achieve to have LAN connection at this point.

softhack007, 2022-10-27:
Another idea - possibly related to flash speed?
We've seen random flickering v0.13.2 and v0.14.0 that disappeared after adding these lines (platformio.ini):
```ini
board_build.f_flash = 80000000L
board_build.flash_mode = qio
```

It could also be the over way around: in case you already use firmware for 80Mhz, then maybe the flash on the WT32-ETH01 does not run reliably at 80mhz, and building for 40Mhz might help.

zigomatichub, 2022-10-28:
The WT32-ETH01 use a chip WT32-S1 where in the datasheet, I found this: On-board clock 40 MHz crystal.
I don't think it's the ESP speed but may be it's a limitation.
I will continue some research on that with the 2 parameters mentionned and test multiple firmware. I will take some days.

zigomatichub, 2022-11-03:
I tried with this compile option but same issue. Working fine at configuration time but not after the reboot.

```
[env:esp32_eth]
board = esp32-poe
platform = ${esp32.platform}
platform_packages = ${esp32.platform_packages}
upload_speed = 921600
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32} -D WLED_RELEASE_NAME=ESP32_Ethernet -D RLYPIN=-1 -D WLED_USE_ETHERNET -D BTNPIN=-1 -D WLED_DISABLE_BLYNK -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_ALEXA -D WLED_DISABLE_INFRARED -D WLED_DEBUG -D USERMOD_MULTI_RELAY -D USERMOD_AUDIOREACTIVE
lib_deps = 
 ${esp32.lib_deps}
 https://github.com/blazoncek/arduinoFFT.git
board_build.partitions = ${esp32.default_partitions}
board_build.f_flash = 80000000L
board_build.flash_mode = qio
```

zigomatichub, 2022-11-08:
Today I compiled again with the main branch (71d84fec) with same configuration but I erase completely the flash + esp32_bootloader_v4.bin. 
I get very positive result now. With ws281x, it works. When I change to APA102, save button then it's nok. After a reboot, it never goes to LAN but use wifi instead.  
In other to have again running with WS281x , I have to disconnect the cable , removed the Lan configuration, and then put back the Lan configuration. 
Do you think writing APA config erase something on LAN config ?

softhack007, 2022-11-09:
> When I change to APA102, save button then it's nok.
Could be a hardware conflict between APA102 and LAN driver, for example:
* GPIO you selected for APA102 are needed also for the LAN interface.
* conflicts in SPI driver, like vspi vs hspi?
It could explain your observation that it's not working after reboot, if LED drivers are initialized before the LAN driver.

zigomatichub, 2022-11-09:
I tried reversing clk io2  / data io4  --> Nok as well.

APA102 code do something that LAN don't like but I would not think about GPIO allocation issue which are already used by LAN :)

nwalke, 2022-11-11:
I'm running a QuinLED-ESP32 ethernet board and experiencing the same (guess I'm not sure if the chip that uses is LAN8720 or not) with APA102 strips breaking ethernet communication.

BlackBird77, 2022-11-11:
I think it has something to do with the SPI initialization over the Arduino
framework! This is the standard implementation in WLED for SPI!

If you compile it with the SPI DMA driver it works on the same Pins..

I have looked for differences in the SPI initialization and code
generally..

The only difference I can see is, the Arduino implementation do not
deactivate the SPI Quad Pins..

The Esp32 SPI DMA does a explizit deactivation of the SPI Quad pins..

On my Olimex Board the IOMUX SPI Quad Pins are on the same pins with the
LAN Chip!

Greetings!
Andre

benkuper, 2023-03-11:
Same problem here, did anybody find a solution ?

BlackBird77, 2023-03-11:
Yes compile it yourself and use the DMA Drivers for SPI..

This works perfectly!

Greetings

blazoncek, 2023-03-11:
@BlackBird77 why don't you make a PR for your findings?
Please.

benkuper, 2023-03-11:
DMA Driver for SPI, looked in few places and didn't find any clear way of doing it for WLED... any pointers on where to look ?
Thank you

BlackBird77, 2023-03-11:
It is only configuration..
Nothing complicated.
You have to disable the standard SPI driver and use the DMA SPI driver.

But this is only a workaround, not a fix for the real problem, of the
original SPI driver.

blazoncek, 2023-03-11:
> It is only configuration.. Nothing complicated. You have to disable the standard SPI driver and use the DMA SPI driver.

Not everyone knows everything. Please.

benkuper, 2023-03-11:
Ok thanks, sorry if I was not clear by saying that searching led me to nothing, and as a developer for software that people keeping asking "dumb" questions, I know how this will feel but still :
Would it be possible to give a bit more information, like file to modify and some lines of code ? Or a link to whoever would have already done that.

Thank you very much

BlackBird77, 2023-03-11:
Replace bus_wrapper.h lines
``` C
//APA102
#define B_HS_DOT_3 NeoPixelBrightnessBus<DotStarBgrFeature, DotStarSpi5MhzMethod> //hardware SPI
#define B_SS_DOT_3 NeoPixelBrightnessBus<DotStarBgrFeature, DotStarMethod>    //soft SPI
```
with
``` C
//APA102
#define B_HS_DOT_3 NeoPixelBrightnessBus<DotStarBgrFeature, DotStarEsp32DmaVspi5MhzMethod> //hardware SPI with DMA
#define B_SS_DOT_3 NeoPixelBrightnessBus<DotStarBgrFeature, DotStarMethod>    //soft SPI
```

This change forces the NeoPixel library to NOT use the Arduino SPI implementation. I don't know why this makes a difference. I searched also all the code. And compared the difference in ARDUINO and ESP32 DMA SPI implementation. All i found was, Arduino do not initialize the QUAD SPI pins.. The DMA SPI Driver explicitly disables the QUAD SPI pins.. And i found the default QUAD SPI PINS are on ports where the Ethernet chip is connected. 

I hope this helps.

Andre

blazoncek, 2023-03-11:
Aha!
WLED is coded to use HSPI peripheral, not VSPI. Please check if other places are broken (usermods) when using VSPI for LEDs.

Thank you.

blazoncek, 2023-03-11:
Will add this to main when I have time to test it a bit:
```
#ifdef WLED_USE_ETHERNET
#define B_HS_DOT_3 NeoPixelBrightnessBus<DotStarBgrFeature, DotStarEsp32DmaVspi5MhzMethod> //hardware VSPI with DMA
#else
#define B_HS_DOT_3 NeoPixelBrightnessBus<DotStarBgrFeature, DotStarSpi5MhzMethod> //hardware HSPI
#endif
```

blazoncek, 2023-03-12:
Please test recent commit. It works on my test system with @conradopoole Ethernet shield and @srg74 universal shield/controller.

benkuper, 2023-03-12:
Tested and working on Olimex ESP32-POE-ISO, connecting SK9822 (APA102 clones) to pins 32 & 33 (same as ethernet), and using SD card MMC at the same time !

Thank you all very much !

AVRnet, 2023-03-12:
Confirm that everything is fine now,
tested and working on WT32-ETH01 with 300 pieces of APA102 to DATA GPIO2 and CLOCK GPIO4.
(WLED v0.14.0-b2)

Thanks to all !!!

blazoncek, 2023-03-12:
Please close rhen.

closed 2023-03-12 by AVRnet