wled/WLED #2544: Alexa discovery issue

sn1ce, 2022-02-14:
### What happened?

Hello!

I am using WLED for a long time now. Recently, I just create a completely new device and used version 0.13 of WLED.

Configured everything, Set-up Alexa in the Sync-Settings, used a custom name ("testlight") and restarted the device, after saving those settings.

But Alexa didn't find the device, I checked all Settings, so everything has a correct setting (Like all my other devices with version 0.12).

### To Reproduce Bug

I downgraded on this device (Wemos D1 Mini) to 0.12 - Alexa instantly works without any Issues.

### Expected Behavior

Alexa should discover the ESP-Controller instantly

### Install Method

Binary from WLED.me

### What version of WLED?

WLED version 0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
---
```

### Anything else?

When I use the 0.12 (with working and connected Alexa) and try to update manually over OTA update to 0.13-b6, it is working, just tested it a few minutes ago.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fhb, 2023-02-14:
still not working with 0.13.3

javigmsuperka, 2023-09-13:
same here , i cannot sync with alexa....

closed 2024-04-23 by github-actions[bot]