wled/WLED #2546: [feature request] Apply most recent time-controlled preset at boot

Jzege, 2022-02-16:
WLED checks every minute for a matching time-controlled preset.
However, I don't leave my LED strip turned on 24/7.
For example, if I set a timer at 6pm, but only power up the LED strip at say 6.30pm, the LED strip will not turn on until the next night, when I really would like it to turn on immediately.

I would like to suggest adding an option in the GUI:
[image]

And modifying checkTimers() like so:
```
void checkTimers()
{
  if (lastTimerMinute != minute(localTime)) //only check once a new minute begins
  {
    lastTimerMinute = minute(localTime);
    int time_delta = 1440;  // minutes in a day
    int16_t nextPreset = currentPreset;
    for (uint8_t i = 0; i < 8; i++)
    {
      if (timerMacro[i] != 0
          && (timerWeekday[i] & 0x01) //timer is enabled
          && timerWeekday[i] >> weekdayMondayFirst() & 0x01  //timer should activate at current day of week
          && (timerHours[i] <= hour(localTime))
          && ((60*(hour(localTime) - timerHours[i]) + minute(localTime) - timerMinutes[i]) <= time_delta) // find the most recent timer compared to localTime
          )
      {
          time_delta = 60*(hour(localTime) - timerHours[i]) + minute(localTime) - timerMinutes[i];
          nextPreset = timerMacro[i];   
      }
    }
  if (nextPreset != currentPreset) applyPreset(nextPreset);
  }
}

```

I would take a stab at it, except I'm still on v0.12.0 and the main branch doesn't work for me.

Aircoookie, 2022-02-16:
Thank you for the suggestion!
A similar functionality has been requested in the forum as well recently.
It shouldn't be too hard to add - just figure out the preset that would have most recently been applied once you have a valid time sync (commonly via NTP, but RTC or UI are also possible) and apply it if the option is set.
Not the highest priority for me, but I'll see when I can get it added :)

blazoncek, 2022-02-17:
Do not forget that, for example, a playlist may be scheduled at 6:00 PM and play only for 15 min. Then stop and not do anything.
In such case starting a playlist at 6:30 PM may be unwanted.

jmerc77, 2023-01-01:
> Do not forget that, for example, a playlist may be scheduled at 6:00 PM and play only for 15 min. Then stop and not do anything. In such case starting a playlist at 6:30 PM may be unwanted.

maybe you do want that but only if "repeat indefinitely" is checked. so no do-overs. maybe even start at the correct preset within (in case of brief power glitch).

blazoncek, 2023-01-01:
Playlist content only exists in `presets.json` it is not feasible to read `presets.json` to determine how playlist behaves.