wled/WLED #2551: Beta 0.13-b6 very unstable

poelzi, 2022-02-19:
### What happened?

I have hard crashes of ESP32 when the playlist is running. Even sometimes corruption of the presets file while not changing anything. 0.12 is much more stable on the same controller. Earlier betas of 0.13 seem more stable then b6

### To Reproduce Bug

Config and presets file. Crash happens after some time
https://gist.github.com/poelzi/ba7486eacf3480ec5b1489ddb9743d2d

### Expected Behavior

No Crash

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-02-21:
If you can, please build with -D WLED_DEBUG and monitor serial output from WLED.
Once you see a crash, please paste that output here so we can investigate.

olemr, 2022-02-22:
Just created my first playlist, and also noticed the crash when looking at the Uptime in Info.
Playlist works 1 time when saving it. Additional run crashes.

These are my 2 presets involved:
[image]
`{"on":true,"bri":232,"transition":20,"mainseg":0,"seg":[{"id":0,"start":0,"stop":67,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 67","col":[[255,0,0,65],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":1,"start":67,"stop":71,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":2,"start":71,"stop":79,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Blå 8","col":[[0,0,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":3,"start":79,"stop":83,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":4,"start":83,"stop":156,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 73","col":[[255,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":5,"start":156,"stop":160,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":6,"start":160,"stop":168,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Blå 8","col":[[0,0,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":7,"start":168,"stop":172,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":8,"start":172,"stop":217,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 45","col":[[255,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":9,"start":217,"stop":221,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":10,"start":221,"stop":229,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Blå 8","col":[[0,0,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":11,"start":229,"stop":233,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":12,"start":233,"stop":278,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 45","col":[[255,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":13,"start":278,"stop":282,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":14,"start":282,"stop":290,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Blå 8","col":[[0,0,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":15,"start":290,"stop":294,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,255,255,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"id":16,"start":294,"stop":300,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 6","col":[[255,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":1,"sx":234,"ix":126,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
`

[image]
`{"on":true,"bri":21,"transition":20,"mainseg":0,"seg":[{"id":0,"start":0,"stop":67,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 67","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":1,"start":67,"stop":71,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":2,"start":71,"stop":79,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Blå 8","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":3,"start":79,"stop":83,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":4,"start":83,"stop":156,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 73","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":5,"start":156,"stop":160,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Hvit 4","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":6,"start":160,"stop":168,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Blå 8","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":7,"start":168,"stop":172,"grp":1,"sp [... cut ...] 255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":16,"start":294,"stop":300,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"n":"Rød 6","col":[[255,191,129,65],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
`
[image]

[image]

blazoncek, 2022-02-23:
As said: please get a debug build, hook up serial and monitor serial output.
When crash occurs, copy and paste that output here.

olemr, 2022-02-23:
I heard you, but I thought as a developer that you would be interested in a 3 step reproduction procedure so you could see for yourself and shorten the round trip test loop when trying to fix it. Guess I was wrong. Sorry.

blazoncek, 2022-02-23:
You were, since it doesn't crash on my end. Sorry to disappoint you.

olemr, 2022-02-23:
I'm in no way disappointed, just didn't know you were unable to reproduce the crash. It is after all a Beta.

blazoncek, 2022-02-23:
@olemr I had the time to play with your presets and playlist and while I couldn't reproduce reboot I did notice one odd WS behaviour.
Please open web browser's developer tools and observe network tab and report back.

BTW I know we all love our languages but try to use ASCII characters in preset/segment names. This can help disgnosing problems.

poelzi, 2022-02-24:
I got the first hard lock with this as the last lines:

```
---DEBUG INFO---
Runtime: 13052705
Unix time: 1645744166,705
Free heap: 201252
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 172.20.73.168
Loops/sec: 1913
Max UM time[ms]: 1
JSON buffer size: 9749
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer released. (9)
Publish MQTT
JSON buffer released. (9)
Publish MQTT
```

blazoncek, 2022-02-25:
No WDT reset? Just lock?

I have this in my source in MQTT code:
```c++
  char apires[1024];                        // allocating 1024 bytes from stack can be risky
```
try disabling (removing config) MQTT to see if MQTT is causing issues.

poelzi, 2022-02-25:
Thats the first thing I tried, unfortunately, not the culprit.

```
---DEBUG INFO---
Runtime: 29105649
Unix time: 29105,649
Free heap: 203352
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 172.20.73.168
Loops/sec: 1860
Max UM time[ms]: 1
JSON buffer released. (9)
JSON buffer size: 4381
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer released. (9)
JSON buffer released. (9)
```

olemr, 2022-02-25:
I got rid of all 'ø' & 'å' s, but it still crashes on the 1-4th run.
This is the data from a Playlist Preset that crashes:
[192.168.1.46.zip](https://github.com/Aircoookie/WLED/files/8144804/192.168.1.46.zip)

olemr, 2022-02-26:
Absolutely. 
Loading a binary is a breeze. It was the compiling bit that I wasn't too sure about. Never done it.
I'm using this pre-assembled 'QUINLED DIG UNO V3 DIGITAL LED CONTROLLER !NEW WITH ESP32'
How do we connect to the serial port?
If through the USB connector, I might need to run without this fuse, but that shouldn't be a problem?
[image]

blazoncek, 2022-02-26:
You don't need serial for uploading firmware but you need it for monitoring debug output.
You can just unplug ESP from Dig-Uno and connect it to USB. Make sure you have serial drivers installed.

Please continue this discussion on Discord where I will also provide debug build.

olemr, 2022-02-26:
After a fruitful Discord discussion, this procedure fixed the crash issue for me. (v2 & v3 Qdiguno)
1. Check you info. If the file system total size is ~1500kb, you are on the old flash layout.
[image]
2. Backup your preset and config.
3. Start the WEB flasher: https://install.wled.me/ Select the latest version.
4. Click install. A COMport selection window appears.
5. Connect your ESP with USB. COMport window should update with a new entry. Choose that.
6. After update, restore presets and config.
Filesystem should now be ~980kb
[image]

I found that the config restore didn't work for me, so I had to specify the LED setup once more.
The LED1 output is GPIO16!

My playlists now works as expected.
Thanks to blazoncek for great support.

blazoncek, 2022-02-27:
For config restore you need to upload cfg.json (either from Security & Updates page or /edit) and reboot immediately.

poelzi, 2022-02-27:
Unfortunatelly, seems not my bug. I have also 983kb in the filesystem. Will test blazonceks debug build now

poelzi, 2022-02-27:
I can't see any discussion with special debug builds in discourse. please give a link

blazoncek, 2022-02-28:
Please join Discord and PM me.

blazoncek, 2022-04-16:
Closing since 0.13.1 has been released.

closed 2022-04-16 by blazoncek