wled/WLED #2553: Incorporate dimming feature Elekstube_IPS

RedNax67, 2022-02-20:
This feature, Night time mode: dims displays and backlight during night time hours, as found in  https://github.com/aly-fly/EleksTubeHAX would be a great addition to WLED.

It has implemented dimming in tfts.cpp:

        r *= dimming;
        g *= dimming;
        b *= dimming;
        r = r >> 8;
        g = g >> 8;
        b = b >> 8;

Aircoookie, 2022-02-20:
Good idea :)
LED Backlight dimming is already fully implemented, as those are just regular WS2812 LEDs :)
Dimming display contents is only partially effective, as the display IPS backlight glow is still quite bright. The "correct" way to do it would be to send a PWM signal to the display backlight pin, which allows dimming without affecting the displayed colors and also reduces IPS glow at lower brightness.
Unfortunately I don't think this pin is connected to the ESP32 in the EleksTube hardware, so only a hard on/off with the TFT_ENABLE pin is possible.
Still I might add this as it will be effective at least when using a bright clockface.

RedNax67, 2022-02-22:
The hardware design doesn't allow for pwm-ing the backlight. So this is a rather elegant solution. Quote from https://github.com/SmittyHalibut/EleksTubeHAX

>  "Tied to system ground through a MOSFET, controlled by ESP32 pin 12, IO27, so the displays can be completely turned off.
> (If I (SmittyHalibut) were doing this, I'd have used a P channel MOSFET and controlled LEDA, which would allow dimming as well as completely shutting it off. Oh well.)"

Submitted a PR with dimming (and .clk) support

Aircoookie, 2022-03-09:
Closing as PR is merged :)

closed 2022-03-09 by Aircoookie