wled/WLED #2556: Preset-/playlist start on different WLEDs with one click on a master.

Bebick, 2022-02-23:
Hey,
I have been using several WLEDs for a long time.
On these devices I have different effects and color palettes on different presets and these in turn stored in playlists. 
Currently (to my knowledge) it is not possible to start the playlists of the different WLEDs directly via a WLED master. Is this function available somewhere or would this be an idea for an extension?
If yes then I would suggest to add another checkbox on the sync setting page where only the number of the preset is transmitted.
So you get the possibility with one click to start the different playlists.
[image]

blazoncek, 2022-02-23:
It has been suggested several times but unfortunately is not technically viable.

Aircoookie, 2022-02-23:
Hey,
while I like the basic idea of preset syncing, I do not know how it would best be accomplished in this case, since presets are per-instance and the same preset ID may correspond to a different effect or even no preset at all.

Bebick, 2022-02-23:
Thank you for your answers.
In my mind I was thinking that you just send the preset number or just send a "simple" start command to all devices and they will play the presets / playlists. Not all different IDs (with effect, color etc) should be sent.
Similar functionality as with the start preset, only that this should not be played after the start, but after a command from the master.
As soon as the button is switched on, one has no more possibility to send effects, color, brightness etc. but only one sends the start of the already preconfigured presets of the other devices.

blazoncek, 2022-04-16:
IMO this may not be feasible or necessary. Syncing presets has no real added value unless it is _very_ specific (implementation and/or installation).

SkylarTheGrey, 2022-05-08:
I'm not sure how this would be so specific an implementation? I too would love this feature either by adding a sync preset setting that sends preset # start command to sync group (ignore if preset # not defined). I also had the thought that this function could be added by allowing API macros in presets to have a "Send to Sync Group" option which would be the more flexible approach and could be useful for more than just this use case. I would also have a use for preset trigger over E.131 to handle segments, though I understand a segment improvement for E.131 is in process.

I do realize that I could build my own code to use the API that handles all this, but that is significantly less maintainable and only solves my install, not the next person to try this. I'm also not remotely a good enough coder to handle the dynamic node acquire, AP generation, DDNS, DHCP, etc. that WLED AP mode nicely handles. (Cherry on top, I install systems and have limited access to maintain, so the more baked in to WLED stuff is the better for maintenance by whomever gets to run the install daily.)

I think that many people may want multiple WLED devices and here are a few reasons that come to mind:
1. they may not all have same chips ad need color adjustment per device to match in physical space
2. may have specific segment designs that only a local preset can properly handle bounds (eg. bounds move in each scene)
3. intentionally want complementary colors or effects rather than matching device outputs. 

Syncing scenes rather than syncing identical color/effects seems like a logical next step to me in creative output of this project.

I get that DDP could do this (except I have 7 devices, 1 is 1300 pix, 3 are 1000 pix each, 3 are 340 pix and usually want different colors on each one and that is asking too much of one central brain). I even managed to do this without DDP but using segments and custom start indexes though again the central brain got overloaded.

SkylarTheGrey, 2022-06-09:
> Hey, while I like the basic idea of preset syncing, I do not know how it would best be accomplished in this case, since presets are per-instance and the same preset ID may correspond to a different effect or even no preset at all.

For me this would be exactly what I want, we need a way to better set each ESP32 to different colors/effects/speeds with an easy single trigger method (button/relay on a 'server' ESP32, E.131, etc) We are a theater that wants complementary effects, not always identical. We do not have the network or address space to run RGB mode, effect mode is close but looses tertiary color and (hopefully soon improved) segment options. 

Ideally for us, if a preset exists it loads, if not then it is ignored. We would likely have some devices that do not trigger in every scene.

blazoncek, 2022-06-09:
Use external software like xLights, Jinx or similar.
UDP WLED relies on is unreliable.

Bebick, 2022-06-09:
I am not quite sure if I have explained everything correctly.
The advantage of simply passing the preset number is that you can start different WLEDs with different presets/playlists at the same time. So you can achieve a synchronicity and you don't have to control every single WLED individually.

I don't know the effort but is this really a big change / extension?

blazoncek, 2022-06-09:
Again: UDP over WiFi is very unreliable. A packet gets lost quite often even in very good environments where WiFi signal strength is adequate. This has nothing to do with WLED.
Even WLED's countermeasure to send notification twice is often not enough (though it may help).

Syncing preset IDs has only added value in specific set-ups but require plenty of work on WLED side (keeping it backward compatible) and is not feasible for us. If you are willing to implement it, by all means do it and submit PR.

mxklb, 2022-11-20:
Maybe someone could test the "new" DMX `Preset` trigger to sync presets via E1.31/Art-Net https://github.com/Aircoookie/WLED/pull/2891

It controls presets for all individual WLED devices via [DMX](https://kno.wled.ge/interfaces/e1.31-dmx/). This should be able to somehow solve this issue.

508loopdetect, 2023-11-29:
you can call preset or playlist names in home assistant for each wled module name recognized. 

so there is a webhook, API Call or similar possible from outside, but not possible from one WLED to the other?

j3rm, 2024-09-22:
Just to throw another use case on this. I have a use case where I want to create a playlist of effects that are different for 4 WLED controllers. 
I'm trying to sync them all to timing of song for presentation. Each does its part at a specific point in the song. 

If I built 4 different playlist, one on each controller with the same id 100, but different order of effects and timing. It would be nice if WLED using esp now could broadcast start playlist 100 to everyone RECEIVING ch 3 from the master controller. This would allow me to start playlist 100 on the master controller and the other 4 nodes would do their own 100 playlist, playing out the different playlist on each

ESP now works really well with sending all following nodes the presets and plays a playlist that it has, but if we could have it broadcast start playlist 100, that would open some very cool ideas and not even require WiFi, api, etc.