wled/WLED #2557: wled json api over serial communication problems

miwied, 2022-02-23:
Hello guys and especially @Aircoookie !

### my setup:
I have a "main mcu" (which is a esp32) that shall control the "wled mcu" (esp8266 d1 mini) via serial communication.
I use the WLED-0.13.0-b6 version. I compiled it myself and changed the Serial baudrate to 921600, hoping to improve serial communication.

### current state:
I experimented with the json api over serial now for about 5 days, and after some success I stuck on an annoying problem / (bug?).
the communication itself works but not **ALL** JSON objects arrive at the wled mcu.
I've read about a "verboseResponse mode" which sometimes blocks the serial input, but I think that this is not the case with my setup.

### the problem:
Unfortunately, despite the successful sending of a json packet, the expected change often does not occur.

### demo / test code
Here you can see a very simple demo code which I wrote for sending the JSON objects:
[https://github.com/miwied/wled-json-api-over-serial](url)

Do u guys have some ideas what the problem could be and how I could fix it?
I've already had read #1173 and #1146 but unfortunately it didn't help me.

(I'm sorry if there are grammatical errors, I'm German)

miwied, 2022-02-23:
Thanks for the fast answer!
Indeed I was using this tool for this sketch.

If I copy ur JSON object 
>{"on":true,"bri":110,"seg":[{"col":[[255,255,255]]}]}
and paste it into the assistant, the code is exactly the same as I used.

```
StaticJsonDocument<192> doc;

doc["on"] = true;
doc["bri"] = 110;

JsonArray seg_0_col_0 = doc["seg"][0]["col"].createNestedArray();
seg_0_col_0.add(255);
seg_0_col_0.add(255);
seg_0_col_0.add(255);

serializeJson(doc, output);

```

miwied, 2022-03-08:
### Problem solved!
I can't exactly say why, but when I used the HardwareSerial instead of the SoftwareSerial, the communication worked without any problems and without any input lag. For me the problem is solved :)

Aircoookie, 2022-03-08:
Nice! SoftwareSerial is only suitable for very slow baud rates, definitely not 921600 in most cases. You should always use HW serial if possible, and in case SoftwareSerial is indispensable, use the lowest possible rate.