wled/WLED #2558: APA102 preconfigured with too high clock frequency

itsinn, 2022-02-24:
For me it was unclear why APA102 worked flawlessly with ESP8266 but had issue after some LEDs with ESP32.
Have realized a difference between the configure clock frequency.

Per default 10MHz are used for ESP32 which doesn't make sense since APA102 have the high frequency issue after a certain amount of LEDs.

I solved it via a hack in DotStarGenericMethod.h

typedef DotStarSpi**2Mhz**Method DotStarSpiMethod;
typedef DotStarSpi**2Mhz**Method DotStarVspiMethod;
typedef DotStarHspi**2Mhz**Method DotStarHspiMethod;

Is there a possibilitiy to integrate a configurable item in the HTTP GUI?

blazoncek, 2022-02-25:
You can try it yourself by adding option in settings_led.htm and appropriate call in bus_manager.h
If you are unable to pull it off, just follow sample for WS2801 to make it compile time option.

Aircoookie, 2022-02-25:
@embedded-creations has also implemented this and it will be merged to master and settable in Ui soon

blazoncek, 2022-02-25:
> @embedded-creations has also implemented this and it will be merged to master and settable in Ui soon

I thought that was stale since there weren't any updates on that. It would also be good to enhance it for WS2801.

Imperattor, 2022-03-08:
> For me it was unclear why APA102 worked flawlessly with ESP8266 but had issue after some LEDs with ESP32. Have realized a difference between the configure clock frequency.
>
> Per default 10MHz are [...]

Am I correct in assuming that this parameter affects the refresh rate of the LEDs? 
I'm having this problem with the refresh rate. The LEDs flicker in the frame. 
Watch video [Watch video ](https://youtu.be/d7Lfb1PoWS4) 
And with both 2812 (800Hz), and 2813 (2kHz), although they have different frequencies.
Maybe you can tell me where to set this frequency, so that they do not flicker on the camera?

blazoncek, 2022-03-08:
> Am I correct in assuming that this parameter affects the refresh rate of the LEDs? I'm having this problem with the refresh rate.

No, this (issue) has nothing to do with your problem. Have you tried APA102 LEDs for your project?
It may be worth trying since APA102 were designed with very high update speed in mind (hence higher PWM frequency probably).

Imperattor, 2022-03-13:
> No, this (issue) has nothing to do with your problem. Have you tried APA102 LEDs for your project? It may be worth trying since APA102 were designed with very high update speed in mind (hence higher [...]

it looks like I already answered you on the forum about apa102
[video 1](https://youtu.be/h-z9AmBu7n0)
[video 2 ](https://youtu.be/z6uqJ23a-XU)

Aircoookie, 2022-03-14:
[This](https://github.com/Makuna/NeoPixelBus/pull/441#issue-817321755) leads me to believe that 5Mhz might be the sweet spot between speed and reliability. I've changed the default to that value. @itsinn I would be grateful if you tested if latest master works for your application :)