wled/WLED #2563: How do I get rid of the flicker on the camera?

Imperattor, 2022-02-27:
there is always flicker when taking pictures on the camera (phone or pro).
[watch the video](https://youtu.be/d7Lfb1PoWS4)

I have tried options: 
ws2812+esp01
ws2812+esp12
ws2813+esp12
There is flicker everywhere
Please help or tell me where to look

Aircoookie, 2022-02-27:
As the PWM frequency of the LEDs is not adjustable, you'll need to adjust the shutter speed of your camera.

Imperattor, 2022-02-27:
> As the PWM frequency of the LEDs is not adjustable, you'll need to adjust the shutter speed of your camera.

It's not about the shutter frequency of the camera. 
On the concert stage there are many LED screens and their frequency is much higher and there is no flicker when shooting. 
The principle is the same - the refresh rate. For example ws2815 supports refresh rate up to 2GHz. and i thought, if the chip supports it then it is possible to set the software to that frequency?

Aircoookie, 2022-02-27:
That's right - the higher the PWM frequency of the LEDs, the less likely it is that this causes visible flicker on commonly used shutter speeds.
From the datasheet, it seems like WS2812(B) uses a frequency of 400Hz, while WS2813 and WS2815 use 2kHz. 
Therefore I am surprised you are seeing the issue to the same degree on both chipsets.

In any case this frequency is a fixed aspect of the used hardware, and there is nothing we can do from the software side to change it.
The only workarounds are using a slower/longer camera shutter speed, using a chipset with higher refresh rate (WS2813 should be ok, so I'm not sure why you are observing the problem with that to the same degree) or driving the RGB channels at 100% (which is not feasible in most cases)

Imperattor, 2022-02-27:
> That's right - the higher the PWM frequency of the LEDs, the less likely it is that this causes visible flicker on commonly used shutter speeds. From the datasheet, it seems like WS2812(B) uses a [...]

Exactly! The frequency is different, but the flicker is the same. 
I was hoping to get less flicker too, so I installed 2813. But it turned out nothing changed. Hence the assumption that somewhere at the software level this frequency can be increased.

Imperattor, 2022-03-03:
Anybody help! Please!

Imperattor, 2022-04-07:
> Just a hunch, but while the PMW is at a certain frequency, the power supply that is powering the LEDs is connected to the power grid right? Thus possibly subject to interference from the frequency of that, because the normal frequency there would be 50/60hz and would give the same effect as you see when filming tube lights and AC driven LED lights. If you do the same test but power the LEDs with a battery, do you still see the flicker? If no, then a better power supply may be the solution, (or some magic involving capacitors, but that is out of my skill range).

Thanks, of course, but that's the first thing I checked)

fusionstream, 2022-06-27:
You'll need to match frequencies of both the strips and the camera. This is a well known issue (Live Events guy here). Power supply issues (both led and camera), transmission issues (led; as well as the camera signal chain for live events), as well as individual chip (on the led) issues are unavoidable. The simplest solution is to lower your shutter speed till it disappears and hopefully it will still be usable. Watch out for the moire.

closed 2024-04-23 by github-actions[bot]