wled/WLED #2568: KNX Implimentation

Alex0176, 2022-03-05:
**Is your feature request related to a problem? Please describe.**
The featurerequest is primary for use WLED via KNX Bus or better the Presets for KNX Group Addresses. 

**Describe the solution you'd like**
There is a git project for use KNX and can be used for set the KNX Address and send and receive data from/to KNX Bus with a Bus ,,koppler". In the Video you see whats needed for KNX and the wireing:
https://www.youtube.com/watch?v=KtBNI-xF23w

the KNX lib can be downloaded here:
https://knx-user-forum.de/filedata/fetch?id=1549797

- So you can select presets via Group Adresses
- Send % values via KNX (for example for garage doors or water filled or temp 

**Describe alternatives you've considered**
In WLED is needed to:
- Implement the KNX Lib.
- Add a enable KNX with the KNX Adress (this must be the same like a Dummy device in ETS)
- Add a possibility for Group Addresses for sending and receiving the values. 

Values:
Send to KNX: actual preset, on/off WLED (for Powersupply)
Receive from KNX: set preset, set % value (for special effects)

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

Aircoookie, 2022-03-06:
This is a good idea and would be best implemented as a WLED usermod by a knowledgable user with a KNX system :)

Alex0176, 2022-03-06:
i think the parts todo: ( i can read C/C++, but i have no knowlage in this) 
+ add knx lib 
+ add on init file (i dont know where to find) if knx is enabled set the correct physical adress to the knx lib.

if (knx ==true) {
    KnxTpUart knx(&Serial1, "x.x.x"); // "x.x.x" should be the stored physical adress in cfg.cpp
    Serial1.begin(19200, SERIAL_8E1)
    knx.uartReset();
}
+ add the information into the preset.cpp (for save/load the settings)
+ add enable knx and physical adress to cfg.cpp
+ modify the html templates 
+ make a thread loop for the knx part and fire up events on the correct lines of code to knx, if its enabled. 
for example in present.cpp (hopefully thats the correct point) 
... applyPresent(.....)
if(knx ==true && presentGroupAdd!=null){

bool success = knx.groupWriteBool("x/x/x", index);
.....

Alex0176, 2022-05-11:
Today i had 2 hours time and check out the usermod v2 example and i make a few things, but i dont know how i can add config fields to define the group address directly at the presents (for now i make a possibility of 50 presets hardcorded and i dont know how to compile the staff. My atom IDE is sucking very at the autocomplete and dont like the code from wlan in my configuration. 

but i only start atom and check out the wled code and make a modification from that as base of my usermod. 

Can anyone compile it for me or help me with my questions?
[KNX-Presents.zip](https://github.com/Aircoookie/WLED/files/8673492/KNX-Presents.zip)

Alex0176, 2022-06-06:
i got now a link error, but i'am not able to fix this. i think there is a issue with casting the serial port to a stream. 

Can anyone help me in this case? 

```
/../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\src\usermods_list.cpp.o: in function `_GLOBAL__sub_I_physicalAddress':
usermods_list.cpp:(.text.startup._GLOBAL__sub_I_physicalAddress+0x2b): undefined reference to `KnxTpUart::KnxTpUart(Stream*, String)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\d1_mini\firmware.elf] Error 1
```
[KNX-Presents-a0.0.2.zip](https://github.com/Aircoookie/WLED/files/8842591/KNX-Presents-a0.0.2.zip)

blazoncek, 2022-06-06:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

Alex0176, 2023-02-17:
> @Alex0176 Have you made any progress one this?

i dont know why it does not compile ... and noone like to help me in this case.

softhack007, 2023-02-19:
> Today i had 2 hours time and check out the usermod v2 example and i make a few things

Welcome to the world of WLED development :-)
Not to spread frustration, but I'd encourage you to spend more time on the task. 2 hours is surely to little time to even understanding the basic.

----

> #include <KnxTpUart.h>
> ^
> compilation terminated.
> undefined reference to `KnxTpUart::KnxTpUart(Stream*, String)'
collect2.exe: error: ld returned 1 exit status

This looks like platformIO cannot find a library that provides KnxTpUart.h. which library do you want to use? 
This library looks similar to what is missing: https://github.com/majuss/KnxTpUart

----
> add the information into the preset.cpp (for save/load the settings)
> add enable knx and physical adress to cfg.cpp
> modify the html templates

Actually you don't need to fiddle with set.cpp + cfg.cpp and  html templates, if you  want to make you own usermod settings. 

This can be done inside the usermod using the standard JSON API.
https://github.com/Aircoookie/WLED/blob/74156b7ed8a1449ed459baeb7934debb53a19665/usermods/EXAMPLE_v2/usermod_v2_example.h#L187-L197

If you need a full example, take a look at the boblight usermod 
https://github.com/Aircoookie/WLED/blob/74156b7ed8a1449ed459baeb7934debb53a19665/usermods/boblight/boblight.h#L307-L351

Alex0176, 2023-02-19:
> KnxTpUart.h

Hi, the libs are included in the ZIP File, and i dont get this kind of compile error. I get the error `/../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\d1_mini\src\usermods_list.cpp.o: in function `_GLOBAL__sub_I_physicalAddress':
usermods_list.cpp:(.text.startup._GLOBAL__sub_I_physicalAddress+0x2b): undefined reference to `KnxTpUart::KnxTpUart(Stream*, String)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\d1_mini\firmware.elf] Error 1`

So its not a libary problem, more a linker or add cpp problem. i dont know - i am not a c programmer, so its hard for me to figure out whats exacly the problem right now.

softhack007, 2023-02-19:
> So its not a libary problem, more a linker or add cpp problem. i dont know - i am not a c programmer,

Actually it is a library problem, as the linker does not find the library where the class KnxTpUart is implemented. 

This typically happens in platformIO if you manage to feed it with a header file, but did not use the proper way of importing an external library. The usual way to do so, is by using `lib_deps` and pointing to a github repository that provides the library.

Alex0176, 2023-05-14:
i dont know how i can configure this - so i hope anyone other with C knowlege can work on it. i can tast and help with the logic

PaoloTK, 2023-08-15:
> i dont know how i can configure this - so i hope anyone other with C knowlege can work on it. i can tast and help with the logic

I'm working on a KNX TP integration myself. I have switch groups working but that's it for now. I was wondering, other than listening and writing to groups that you would expect from a light (on/off, brightness, effect, etc.) is there any other feature you think makes sense to have?

Alex0176, 2023-08-15:
i think the power on/off for the powersully will be a good idea to have. So you can power the ESP via KNX and switch the powersupply via KNX group. The other effects you find on the typical switch solution, like MDT have on his glas panals. 
https://www.youtube.com/watch?v=cx85Y5eTSHQ

PaoloTK, 2023-08-15:
> i think the power on/off for the powersully will be a good idea to have. So you can power the ESP via KNX and switch the powersupply via KNX group. The other effects you find on the typical switch solution, like MDT have on his glas panals. https://www.youtube.com/watch?v=cx85Y5eTSHQ
> 
> did you have already a usemod or custom firmware?

The power supply should switch on automatically when the light is turned on in WLED, why would you add the relay to a separate switch group?

You can find the usermod code in my repos, WLED/KNX. But it's too early to submit it, needs a lot more work done.

Alex0176, 2023-08-15:
yes, but if you make this with ESP32 than you have to switch the 230V relay, but if you have this already on KNX, you can use this switch. This make sense i think.

PaoloTK, 2023-08-15:
> yes, but if you make this with ESP32 than you have to switch the 230V relay, but if you have this already on KNX, you can use this switch. This make sense i think.

I have already implemented a light feedback group. If you have an actuator that controls the light power supply, add it's switch group to the light feedback group. When you turn the light on in WLED it will send a "TRUE" to the group, turning on the actuator.

Is this what you were looking for?

PaoloTK, 2023-08-18:
@Alex0176 I have made some further progress and added basic instructions on getting my usermod up and running. I want to stress this is super early and there will be bugs, but if you want to try it I've got basic functionality (on/off, absolute and relative dimming) working. You can find the instructions and code here, but you will have to custom compile. I haven't tested the usermod with free and 2-level group style, so it won't probably work unless you use 3-level style.

https://github.com/PaoloTK/WLED/tree/KNX/usermods/KNX

If you have any questions or have troubles getting this up and running, let me know.

KlausHans, 2023-08-19:
Very interesting idea! Tasmota has this feature already, the integration in Wled would be perfect.
@PaoloTK I would like to test but i only have a knx interface, no router. I think i will change that in the future, but for now i can not test it.

PaoloTK, 2023-08-19:
> Very interesting idea! Tasmota has this feature already, the integration in Wled would be perfect. @PaoloTK I would like to test but i only have a knx interface, no router. I think i will change that in the future, but for now i can not test it. Edit: i just saw that you dont use Knx Ip.

The interface will work fine too, the WLED devices communicate over twisted pair.

michael-blue2, 2024-03-18:
> @Alex0176 I have made some further progress and added basic instructions on getting my usermod up and running. I want to stress this is super early and there will be bugs, but if you want to try it [...]

Hey @PaoloTK I have been trying to compile your code to test it. 
But I do always get an error that (PackageException: Package version 2.2.0+sha.a68a18c doesn't satisfy requirements ~2.0.7 based on PackageMetadata) 
Any chance you can share the compiled file (.bin file)? 
And have you made any process from the first time you uploaded your code? as its been 7 months.

michael-blue2, 2024-03-18:
I got around that error and just compiled the bin file. I will start right now the test with the bus system. I will be using some abb bus couplers ( a bit over priced) but I already had them around.

PaoloTK, 2024-03-22:
> Hey @PaoloTK I have been trying to compile your code to test it. But I do always get an error that (PackageException: Package version 2.2.0+sha.a68a18c doesn't satisfy requirements ~2.0.7 based on [...]

Hi, glad you got it compiled, let me know how it works. As for progress on the code, I got stuck due to performance issues when trying to support 600+ groups (all settings for each segment for max amount of segments), so I started a complete rewrite with a more efficient memory handling but got kinda stuck and been busy with work. Not sure if it's worth the effort since I doubt most people use that amount of segments.

Alex0176, 2024-03-22:
i dont get it up and running right now - so i wait for a working image or something like this, where we can configure the knx stuff via webinterface. . . than i will add the knx stuff to my wled installations here.

PaoloTK, 2024-08-10:
> Any News?

Looking to start back development on the feature in the next few days. I think it makes sense to ship an MVP and worry about additional features later.

softhack007, 2024-08-10:
> I think it makes sense to ship an MVP

@PaoloTK sorry I'm not a native speaker - what's an MVP, and what happens when you put it onto a ship?

fdepalo, 2024-09-08:
Hey Paolo, i don't know but i imagine we speak the same language. I developed a new tpuart adapter that is working flawlessly with esphome and i just managed to compile your usermod. Do you need help ?

PaoloTK, 2024-09-08:
> Hey Paolo, i don't know but i imagine we speak the same language. I developed a new tpuart adapter that is working flawlessly with esphome and i just managed to compile your usermod. Do you need help ?

Hey. :) 
I did start working on a new KNX library but it's a lot of work, so was thinking of sticking to the current one for the time being, even if it has some questionable decisions (e.g. only passing group addresses as String). I found the openKNX library to be extremely well written but its license is incompatible with KNX and it's only designed to be used with ETS (can't configure device from WLED GUI).

michael-blue2, 2025-01-31:
Do we have any update on this? 
I have been waiting for some months now to test this.  Just finished my first knx project which I haven't yet installed my wled devices.
My first idea was to connect them through home assistant and expose them from there to knx keypads but I do prefer them directly to knx.

marcel151, 2025-04-20:
Any news? I am planning lighting for our new home and also want to realise a few RGBCCT strips (e.g. garage, stairs). I think WLED is _the_ solution for adressable LED strips. Also nowadays there are components that are somewhat "professional" and not the DIY solutions as before, for example this POE module for DIN rail mount installation:
https://shop.codm.de/en/automation/wled/wled-lan-controller-poe-din-rail-range-out

Unfortunately WLED is lacking is KNX support, which more and more homes in Europe (especially in Germany) are equipped with. KNX is going from a former more industrial automation standard to an also Smart Home automation solution. Since it's a well-used standard, you don't have to worry about end of support in the future. That's why it is the solution we are going for in our home.

I think KNX IP support would be ideal for WLED, since you don't need new hardware for it.

I am not a software developer. But I am able to read, understand and alter existing code and happy to assist in testing (when our home is finished).

u20p17, 2025-08-31:
@PaoloTK: thanks for all your effort! 👍 any news on this topic? would be great if it would also work via knx-net in the future 👌

u20p17, 2025-09-05:
Hi all,
I started to implement a KNX_IP usermod v2 to wled v1.15.1 (https://github.com/u20p17/WLED/tree/release/v0.15.1). The whole integration is still untested in a real KNX system as I am still in the planning phase of my house :P Therefore it would be great if somebody of you could test it and give feedback/make commits with bugfixes. (@Alex0176 could you give it a go? - i can also send you a precompiled firmware for an esp32 with knx-ip usermod)

(I took the https://github.com/envy/esp-knx-ip  library as reference (removed webserver and some stuff) as it has the MIT license...)

fdepalo, 2025-09-07:
> Hi all,
> I started to implement a KNX_IP usermod v2 to wled v1.15.1 (https://github.com/u20p17/WLED/tree/release/v0.15.1). The whole integration is still untested in a real KNX system as I am still [...]

hey @u20p17, if you are interested i developed a knx-tp board for esp32, i don't know how much work can be integrating it. Let me know 🤪

u20p17, 2025-09-07:
@dennnnis99 : I uploaded the first beta version here: https://github.com/u20p17/WLED/releases/tag/v0.15.1_KNXIP_beta1 If you can try it would be great - I expect some bugs in there, as I couldn't test it up to now...

@netmindz : I will make a PR as soon as I get positive feedback from some KNX users that it is working fine.

@fdepalo: I will have to check how much work it might be... Still would be interessing for me as well. Are you in the knx-user-forum.de? Maybe we could write there via PM - I have the same name as on github (u20p17)?

marcel151, 2025-09-07:
@u20p17 Thanks for your efforts trying to get KNX to WLED, hopes are high that we'll get a functional version in the future. I am happy to help when we finish our KNX powered home.
I am also interested in a KNX TP version. If there are efforts to integrate it maybe there is a use case to make KNX-TP ESP32 hardware (thanks @fdepalo). I think this would also be interesting for [@codmpm](https://github.com/codmpm), as he is also developing WLED hardware, which are high quality components.

HoffmannOs, 2025-09-08:
Have Flash on a esp32Wroom to test it, in Usermods KNX Shows up, PA Can change to 0.0.12x, but GA dont can change to another Adress.... click Save and reboot, but only the 1/0/1 will repeat.

HoffmannOs, 2025-09-08:
@u20p17 yes, wireshark dont show up. Have check too now.

u20p17, 2025-09-09:
ok I will check it again. i can see some of the messages in wireshark:

[image]

u20p17, 2025-09-10:
@HoffmannOs : Do you see the KNXIP messages in wireshark from the tasmota devices? If so, can you send me of these message? 

To run my script you would have to copy the code snippet in a .txt file, change the ip address to fit to you laptop/pc ip and save it. after that change the filename to 'test.py' and open the cmd window. navigate to the folder where the .py file is located and type 'python .\test.py' to run the script (of course you need python installed on the pc/laptop...). If I do this I can see the KNXIP messages from the Beta4 in wireshark on my system...

u20p17, 2025-09-10:
beta 5 is out now: https://github.com/u20p17/WLED/releases/tag/v0.15.1_KNXIP_beta5. I found out that the cEMI message code was formatted wrong ( I formatted it like a recieve, but should be send...) therefore the KNX-IP router might have ignored the messages...

@HoffmannOs : can you test beta 5 again and if it still is not working send me some messages recorded with wireshark of the tasmota devices? (or use my python script to log the data from the tasmota devices...) Thanks!

u20p17, 2025-09-13:
@marcel151 : thanks for your feedback and good to know that with knx virtual it seems to work fine (y) You are right that we need a KNX router to be able to recieve the milticast messages - an KNX interface is not working for this use case. I will try ro also set up the knx virtual as you described for the future development :)

Regarding the DPTs: Does anybody have expierience which ones normally get used for RGB or RGBW or RGBCCT strips? at the moment we have separte GAs for each color, bit i am not sure if this is the way to go?

Alex0176, 2025-09-13:
you can install knxd and make a knx router via knxd. 

knxd connect to the knx interface (tunnel mode) and make the router part in it. I used debian and compile knxd by my self. Because the debian 13 knxd packet is broken. 

after you compile it, i think this is the correct paramters in the /etc/knxd.conf 
You find the repository here: https://github.com/knxd/knxd

# configuration for knxd.service
KNXD_OPTS="-e 0.0.1 -E 0.0.2:8 -u /tmp/eib -R -D -T -S -b ipt:xxx.xxx.xxx.xxx"

xxx.xxx.xxx.xxx = ip of knx interface.

HoffmannOs, 2025-09-13:
Now i have it Run, on Bus Monitor in ETS comes all correct on With the PA and Backbone State, in Grey Line 357 is 0.0.1 sending from ETS, is maybe my Problem i have to use these GA 6/0/1 over the Tunnel from The Router from Mainline.

[image]

u20p17, 2025-09-14:
Beta9 is out: [WLED_0.15.1_ESP32_KNX.bin](https://github.com/u20p17/WLED/releases/download/v0.15.1_KNXIP_beta9/WLED_0.15.1_ESP32_KNX.bin)

- Added new GA's for H,S,V, HSV, RGB, and RGBW
- Automatically detects LED bus capability and changes GA In/Out accordingly

@HoffmannOs : If i understood correctly it seems that it is now also working in your knx system?

u20p17, 2025-09-16:
RC1 is out: https://github.com/u20p17/WLED/releases/tag/v0.15.1-KNXIP-rc.1

- added GAout for Internal temperature + Dallas Temp Sensor (needs add. usermod)
- added GA out for Overtemp. for both sensors
- added GAin for time/date/datetime 
- precompiled versions for all quinled boards

HoffmannOs, 2025-09-30:
Hey,
have flash the Rc7, but have Problems with the State, on Tasmota comes as State back on Bus when turn on or off, but with GA Out Wled Power   [-] (DPT 1.001) nothing comes back on my Routerside, on the Multicast Side shows up, can you maybe change the Point as State to test? Tasmota gives a 1.011 State. Brightness back comes to router as normal...

[image]

u20p17, 2025-10-01:
The actual KNX communication will work the same way (DPT1.001 and DPT1.0.11 are 1-bit boolean values), so you should already see the message on the configured GAout. I will change the labeling in the next RC to make it clearer. I only send the GAouts if they changed - could that be your issue?

fdepalo, 2026-01-17:
Hello, sorry for hijacking the thread,

next week i will publish a working usermod for knx-tp, i'm testing it right
now.

Cheers :)

[image]

u20p17, 2026-01-17:
i am currently working on an OpenKNX application where you can control addressable leds like wled, but with the big advantage that you can programm the device via ETS (TP and IP will be available)… still work to do but the first alpha version is already running at different OpenKNX developers ;)

fdepalo, 2026-01-17:
my personal opinion: ets is not an advantage but an handicap ;) better to program in yaml