wled/WLED #2569: PWM RGBW Color Order Bugged

DerWackeldackel, 2022-03-05:
### What happened?

Its not possible to change the Color Order from "RGB" to something else. After pressing save it still stays on the "RGB" order.

### To Reproduce Bug

Just go to LED Preferences and try to change the color order setting

### Expected Behavior

Its gets resetted every time you press on save.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b6 and 0.13.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-03-05:
Setting color order is not supported on PWM busses (swap around the pins instead).
The real bug here is that the color order option is displayed for PWM busses, will fix!

DerWackeldackel, 2022-03-05:
Ah okay! Thank you for your quick answer! 👍

commit 02b08939 (2022-03-06): No color order select on PWM busses (fixes #2569)
closed 2022-03-06 by Aircoookie