wled/WLED #2570: ESP32: Analog LEDs glitch out when used along with APA102s

Joshfindit, 2022-03-07:
### What happened?

When using analog PWM signals from the ESP32, the board is fine for a while but within 5min to 2 hours the LEDs 'glitch' and don't recover until a reboot (or sometimes turning the light off/on through the interface)

Glitching manifests as one or more of:
- Significant random flickering with no pattern
- No light output at all
- Intermittent flickering while still following the timing of the pattern
  - In the "breathe" effect, for example, the light will rise and fall as expected, but intermittently it will also flash full brightness or flicker completely off before resuming the pattern

### To Reproduce Bug

1. Download, compile, and install WLED with the build flag `-D WLED_USE_ANALOG_LEDS` (I did this in `platformio.ini` by adding the flag under 102: `build_flags =`)
2. Wire a few APA102s to 5V, GND, and 2 GPIO pins
3. Wire a 3.3V LED to a separate GPIO pin
4. Boot up the ESP32 and configure WLED with those pins
  - I like to set LED:
    - Output 1 as the APA102
    - Output 2 as `PWM White` (Warm white LED)
    - Make a segment for each output: `true`
5. Turn on both segments at full brightness and set an effect where it's easy to see the glitch (I typically use `Breathe` or `Heartbeat`)
6. Wait

### Expected Behavior

Clean light, and effects that function as defined

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.0-b7 (master)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I'm using the WeMos D1 Mini ESP32 (v3.0.0)

Possibly related to https://github.com/Aircoookie/WLED/issues/1347 (I started another project with a fresh board and quickly ran in to the same issue)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-03-07:
No need for `-D WLED_USE_ANALOG_LEDS` compile time options since 0.12
I do not see the same behavior.

Aircoookie, 2022-03-07:
Are the APA102 or the PWM white LEDs glitching?
It is very possible that if your PWM white light has a significant current, it could cause interference in the data or clock channel of the digital LEDs (although WS281x types are typically more susceptible to this). If possible, make sure the PWMed wire does not run in parallel to the APA102 data and clock wires.
Are you using a level shifter?

Joshfindit, 2022-03-09:
It's the PWM White that's glitching.

It happens on:

- D1 Mini ESP8266
- D1 Mini ESP32
- Quinled-ESP32

Each has similar wiring:

- Power to the board supplied via USB or the 5V pin using a 5V power supply
- For the APA102 + any
  - 5V and ground are connected in parallel to the 5V power (via the 5V pin on the board when powered by USB)
- For the APA102 + ESP32:
  - CLK is connected to GPIO17
  - DATA is connected to GPIO21
- For the PWM LED + ESP32:
  - `+` connected to GPIO22
  - `-` connected to GND on the board
- For the APA102 + ESP8266:
  - GPIO1 (TxD) for clock
  - GPIO2 (D4) for data
- For the PWM LED + ESP8266:
  - `+` connected to GPIO5
  - `-` connected to GND on the board

I did try various resistor values on `PWM LED -` while prototyping, but they made no difference. Currently it has 100ohm.
Since I tested with a single LED (and without any. See below), I don't think PWM current would be a factor.
No logic shifter. I know it's not ideal but I'm working at a very small scale (1-8 APA102s) and most boards show no problems. This is the first time I've combined APA102 and PWM LED however.

Points of interest:
- I measured the voltage on the PWM LED pins after removing the LED itself, and found low voltages on all boards. Usually <2.8V, but it wasn't consistent
- The ESP8266 worked more consistently, but would reliably fail when the brightness was turned up. When it did fail I could reliably fix it by flashing the board, and usually fix it by disconnecting the power and reconnecting it.

Aircoookie, 2022-03-10:
Sounds like you might have a powering issue. What kind of PWM LEDs are you using? If it needs too much current, the ESP might not be able to power it directly and you'd need a transistor/MOSFET.

For ESP8266 specifically, you can also try using GPIO13 for APA102 DATA and 14 for CLK. Those are the hardware pins, which might reduce load on the CPU and make the PWM output more reliable.

Joshfindit, 2022-07-10:
> Sounds like you might have a powering issue. What kind of PWM LEDs are you using?

The LED being used is a 5mm Warm White LED. 20mA of draw at full brightness, and a typical forward voltage of 3.2V.
I’ve used these LEDs for testing ever since I started working with LEDs a couple years ago, and I’ve never had an issue driving them from a single PWM pin.
Do note as well that I put various resistors on that single LED and it made no difference to the pattern of the flickers.

I’m open to the idea of the data lines having interference though I don’t have the skills to make that determination. It would be surprising as I’m using what is now a battle-tested design (D1 Mini v3.0.0 and the ESP32 mini).

I’ve recently skilled up and have a logic analyzer. Is there anything I could do to get hard data on what’s going on so it has a chance of leading to a solution?

closed 2024-08-23 by github-actions[bot]