wled/WLED #2573: NodeMCU blew up

sjude68, 2022-03-07:
### What happened?

I am not sure if it was the problem with the update but let me report it.

I am using Wled for some WS2812 led stip for my showchase.  Everything was working find before the update.  Pin 2 was used for LED strip. I don't remember which pin for relay.  Relay for toggled when stip was powered off as default.  I had used a compiled firmware by manually selecting the relay pin in the code.

Just yesterday I decided to upgrade the firmware to WLED_0.13.0-b7_ESP8266.  As the relay was not toggling as required, I selected pin 2 for relay unknowingly that pin 2 was used for Led strip.  The strip started behaving strangely.  All the presets got reset.  As I had setup 4 segments, I was just doing a selection of 8 leds in 1st segment and clicked the tick to confirm.  That is when the led stip went off and never came on.  The Nodemcu was not connecting to Wifi and nor did the AP mode come on.  The whole setup was fixed behind the wooden panel, I has to dismantel the setup.  I checked the nodemcu and found no powering or blue led coming on. (without any connections on the pins).  I then tried to power the nodemcu by 5V on VIN and GND pin.  Never worked.  Powering on 3.3v by 3.3 v also didn't work.

How can a working nodemcu go off by the beta version.  Just reporting this just incase there is a problem.

### To Reproduce Bug

mentioned above

### Expected Behavior

nodemcu gone faulty.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.0-b7_ESP8266

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
none
```

### Anything else?

none

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-03-07:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

sjude68, 2022-03-07:
What is forum or discord used for?

Forum for discussion and discord for chatting.

I reported a bug. If I use Discord, my comment goes unnoticed to others. Only online members read it and maybe reply. I don't like Discord for this. Once I was searching for a solution but did not find it. Scrolled numerous messages but didn't get the answer. If I post here it can be searched by the topic.

So don't force anyone to post on Discord. That is a online discussion app.

blazoncek, 2022-03-07:
Hardware malfunction, while theoretically possible, is most often not a software issue.
With about 30 devices (some in testing phases some in production) with various setups (ESP01, Wemos D1 mini, ESP32 D1 mini) I have occasionally misconfigured GPIO pins but never _blew_ (or _fried_) any ESPs.

dartfrogdk, 2022-03-08:
Well i got the same problem as sjude68, updated to  WLED_0.13.0-b7_ESP8266 now no contact to esp!

k7bbr, 2022-03-08:
Would appear closing this issue in 11 minutes might have been a little
rash...

blazoncek, 2022-03-08:
Inaccessibility and _fried_ ESP may not be the same thing. It may just be a configuration issue. 😉 
If you all still think it is a bug, please provide reproducible set-up. I am willing to sacrifice a few ESPs if that is the case.

And the issue is not yet closed if you haven't noticed.

dartfrogdk, 2022-03-08:
mine worked like a charm with version 12, then i did an online upgrade to newest version as stated, after that nothing, tried with a usb it cant be connected it is a d1mini, and YES i have done this a million times without any issues.

sjude68, 2022-03-08:
My nodemcu is not emulating the come port. But when I powered by usb and measured voltage on vin and 3.3v ports, I get 5v and 3.3v respectively. So it is not fried. Something else gone wrong with that beta update and pin configuration.

Any help will be appreciated.

sjude68, 2022-03-08:
> Well i got the same problem as sjude68, updated to WLED_0.13.0-b7_ESP8266 now no contact to esp!

So there is somebody who had the same problem. If it is a software issue, it can be recovered. But I am not getting it as a USB device. Any other way to re-flash the firmware.

blazoncek, 2022-03-09:
It may mean your USB to UART chip is dead. That cannot be caused by software.

dartfrogdk, 2022-03-09:
So you tell me that at least 2 chips brake with the same update and it is not the software, hmm my d1mini have been working without problems with many updates and worked after, it updtated to the new version and died when restarting!!!

Aircoookie, 2022-03-09:
@dartfrogdk do you have a separate d1mini that didn't have b7 installed that still work via USB? Just checking your USB cable's data line is fine.

dartfrogdk, 2022-03-09:
yes everything is fine i flashed a new one just now, i also did an update to WLED_0.13.0-b6_ESP8266 on another without problems

Joshfindit, 2022-03-09:
I feel like it might be helpful for someone with one of these to connect it to an FTDI programmer and attempt flash (possibly using a tool like this: https://raw.githubusercontent.com/jimparis/esptool-ftdi/master/esptool-ftdi.py ) or check the console (for example using `screen`)

sjude68, 2022-03-09:
I couln't connect by USB so I reported my nodemcu dead.  But today after reading the messages and @dartfrogdk's comment that USB port cannot go faulty by software, I decided to give a try again.  And yes, it got emulated as COM3 port.  So I flashed version 12.  I have not yet connected any LED strip or relay to check. 

Thanks everyone to keep this chat active.  On Discord I would'nt get a solution as I am in India and my timing to stay online are different from you guys.  Thanks again.

softhack007, 2023-01-12:
This issue
* could be a hardware problem
* has not been reported for any current versions (0.13.0-b7 is more than a year old)
* double use of pins (leds and relay) is clearly a hardware electrical problem, double configuration of pins should be prevented by pinManager.
* had no activity since > 6 month

-> closing this ticket.

closed 2023-01-12 by softhack007