wled/WLED #2575: Issue running latest version WLED on the SP108E

sogeniusio, 2022-03-10:
I double checked my wiring. And I'm assuming there's an issue with the controller. I followed the steps on the WLED compatibility.

I'm seeing a ton of flickering, not sure why. I'm running the latest version of WLED. I've tested the LEDs with a different controller and it works fine. If I remove the jump wires from the controller, I get solid white. 

I posted the video of what I'm seeing on Reddit. Didn't see an option for uploading it here. 

https://www.reddit.com/r/WLED/comments/tb3orh/issue_running_wled_on_the_sp108e/

[image]

dosipod, 2022-03-12:
Hello ,I was looking forward  to flash SP108E and seen this .Did you test with 0.13.0-b6 and it worked or is this the first time you flashed it with wled ? 

In all cases  if you are looking for a quick answer then please try wled  discord  https://discord.gg/HyBAMMeT  , good luck

lolorc, 2022-04-06:
hi,
I'm using a sp108e with ws2812b and infrared.
I've used the information from https://github.com/psxde/sp108e-led-controller

so you're missing 2 wires to keep the extra chip "off" and to put the esp in bootloader mode when pressing the button

it should be noted, on my board it's not GPIO4 it's GPIO12.

Changchung, 2022-04-16:
Hi, sorry for the nob question, but why install wled in this controller? What are the benefits

blazoncek, 2022-04-16:
> Hi, sorry for the nob question, but why install wled in this controller? What are the benefits

Endless possibilities?

blazoncek, 2022-11-18:
Closing due to no updates.