wled/WLED #2576: WLED forces gzipped content, violating HTTP standards.

jamiebegin, 2022-03-12:
### What happened?

The WLED HTTP server ignores the "Accept-encoding" header and always returns gzipped content.

### To Reproduce Bug

Running `curl http://192.168.5.211` will result in a in a bunch of binary gibberish (gzipped compressed content) being dumped to the console. It shouldn't do this because `curl` doesn't tell the web server during the request that gzipped content is allowed.

Trying to explicitly deny a gzipped response using `curl -sH "Accept-encoding: " http://192.168.5.211` results in the same thing.

### Expected Behavior

WLED needs to return a plain, uncompressed response to a request unless the useragent specifically allows a gzipped response.

In my situation, this is preventing me from using Nginx in front of WLED as a SSL proxy. Since WLED has "ws://" hardcoded (bug # #2571), this issue is preventing me from working around the original bug. (The `sub_filter` Nginx module can only rewrite uncompressed content).

### Install Method

Self-Compiled

### What version of WLED?

2202222

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-03-12:
Thank you for reporting this issue!

Serving the uncompressed version is not easily possible, as the server only stores the pre-compressed version by default.
Flash size constraints make it infeasible to store both versions and on-the-fly compression is not an option either to my knowledge.

As a workaround, you can save the index file as .html and upload it to `[WLED-IP]/edit`, that way WLED will serve the page uncompressed.

PR for conditional wss: protocol has been merged :)

blazoncek, 2022-03-13:
Unfortunately you also have to upload iro.js, rangetouch.js, index.css and index.js.

Aircoookie, 2022-03-13:
That many successive requests may reset the ESP. I'd recommend downloading the combined, minified (but due to the download no longer gzipped) source directly from the ESP. We may also want to change cdata.js to output an index_min.htm in addition to the C array output in html_ui.h

blazoncek, 2022-03-13:
I have added throttling logic to my fork a while ago (index.htm) and it works either with uploaded files or compiled in.

blazoncek, 2022-04-16:
I do not think this can be solved without great sacrifices on ESP side.
I deem that unfeasible ATM.