wled/WLED #2586: peek does not work with odd number of led

terbonium, 2022-03-17:
### What happened?

Peek does not work when configured with 285 LEDs.

Works properly with 284 or 286 set.

### To Reproduce Bug

Set led count to odd number, like 285

### Expected Behavior

Expect peek animation to display at top of navigation

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 13.1 Toki Build 2203150

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

CrMo4130, 2022-03-18:
Same here, has been this way since 13b7 for me on ESP32's.

blazoncek, 2022-03-18:
What if you reduce number of LEDs below 255 (and keep odd number)?

Aircoookie, 2022-03-18:
It works with odd numbers <= 255. Taking a look now.

Aircoookie, 2022-03-18:
Ok, fixed in master, although there is now a new bug that causes the last LED of the live view to flicker in random colors. Looking into that now.

commit 86010521 (2022-03-18): Fixed liveview buffer over-write (fixes #2586 )

(with odd LED counts > 256)
closed 2022-03-18 by Aircoookie