wled/WLED #2587: Analog button (potentiometer) causes glitches and flickering

dylan-conlin, 2022-03-17:
### What happened?

I've noticed that attaching at least 1 potentiometer causes my light strip (300 LEDS) to flicker. The flickering seems to get worse if I increase the number of LEDS.

Does this have something to do with frame rate? Is there a workaround or fix?

### To Reproduce Bug

To reproduce, add a potentiometer to control the effect speed.

### Expected Behavior

I expect the lights to not flicker when I add a potentiometer.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.13.0-b7 "Toki" 2202222

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huggy-d1, 2022-03-18:
Which pin is the POT connected to?
Which pin is connected to the LED strip data input?
If those questions have the same answer, there's the problem.

tke233, 2022-04-05:
I'm experiencing this as well, and also if you tuned in to the DrZzs livestream, you see he also had issues with his potentiometer https://youtu.be/M4M7qox2zFk?t=3035 I am using a 20K Ohm potentiometer he was using a 10K Ohm potentiometer.  I am going to try some other valued potentiometers that I have to see if the results differ any.  The board I'm using is a QuinQuad v2. I have 2 potentiometers hooked up (Pin 36 & 32) flicker occurs when using 1 or both.

blazoncek, 2022-04-05:
Try adding yield() right after analogRead() and report back.

tke233, 2022-04-05:
> Try adding yield() right after analogRead() and report back.

I'm on familiar with the point-n-click UI, sorry but if you can give me idiot proof instruction of how to do what you're asking, I will test it out immediately.

blazoncek, 2022-04-05:
Sorry, needs modifying source and recompiling.

shd7899, 2022-05-14:
I noticed the same behavior. In my case there were 4 pots 20K each. Connected to ESP32 GPIO36, 39, 34, and 35. LED's flicker whether you use 1 pot or 4. I tried to use a 0.1uf capacitor with each pin and GND, but it did not help.

blazoncek, 2022-05-15:
Flickering is not related to a capacitor (or the lack of) on the input GPIO.
Try changing the code to wait if `strip.isUpdating()` before processing ADC.

softhack007, 2022-05-18:
Could this be "simply" the noise from ADC? ESP32 has somewhat noisy analogue sampling hardware, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#minimizing-noise

To fight ADC noise, some software filtering might help, like this (just an example to illustrate the idea)
```` C++
static float aRead = 0;
float newReading = analogRead(...);

aRead = aRead + 0.05 * (newReading - aRead);. // exponential filter to smooth noisy input - 0.05 is the filter parameter; adjust to your needs

return(aRead);
````

blazoncek, 2022-05-19:
Noise is somewhat taken care of (by truncating lower bits).

softhack007, 2022-06-13:
POTs are sometimes noisy , causing scratching / glitches when you turn them. Noise from the ADC ( up to 1%) is different, and occurs at any time, also when _not_ turning the POT. 

Both kinds of noise could be reduced by low- pass filtering the value from analogread - see my example above. However truncating lower bit would only help sometimes. Imagine when your value is noisy and dancing araround the value of 15. Now if you truncate the lowest two bits, you still have noise whenever the signal crosses the borderline between truncation "bins". For example: going from 15 to 17 then back to 14.  A simple exponential filter would perform better in this case.

I could try to make a PR, however currently I'm too busy with activities related to WLED-SR. 

Would be great (appreciated, la Ola, winner winner chicken dinner, hero of the minute, etc 😉 ) if someone jumps in to make a PR that includes recommendations from above disussions:
1. wait if strip.isUpdating() before reading a new value from ADC
2. Try adding yield() right after analogRead() - might help or not help
3. Filter the value from analogread- filter parameter tuning needed, so the POT remains responsive but the value changes smoothly

softhack007, 2022-06-13:
I think this would be the code function to look into (@blazoncek please correct me)

https://github.com/Aircoookie/WLED/blob/19c8b4fe2d0d86887d49abd5c965fd1d46502679/wled00/button.cpp#L134

blazoncek, 2022-06-14:
> I think this would be the code function to look into (@blazoncek please correct me)
> 
> https://github.com/Aircoookie/WLED/blob/19c8b4fe2d0d86887d49abd5c965fd1d46502679/wled00/button.cpp#L134

Correct!
It would be great if you could spare a minute or two to make this better. I have never had serious issues while testing potentiometer but I do admit that I do not use it in any production unit.
One thing I did notice is that analogRead() can be quite slow on ESP8266.

softhack007, 2022-06-20:
done - see PR #2693.

closed 2022-06-22 by blazoncek

softhack007, 2022-06-22:
Just for explanation: the PR should solve problems (flickering) related to analog input. It filters out noise, and ensures that effect parameters are not changed while a "strip update" is in progress.

There are other possible sources of "flickering", which are not specific to analog input; when using web UI you might see "flicker-like" interruptions in some effect:
* when changing effect speed or intensity 
* when the brightness is very low