wled/WLED #2589: /json/cfg not working

computergeek1507, 2022-03-19:
### What happened?

In xLights I wrote code to post the LED string configurations to the /json/cfg endpoint. This working in build 2105110 but now in build 2203142(v13.0) it does not.

### To Reproduce Bug

I upload the following JSON using the CURL library.  I have the content-type set to application/json. When I go to the LED Preferences page, none of the settings change.
[wled_json.txt](https://github.com/Aircoookie/WLED/files/8308166/wled_json.txt)

### Expected Behavior

The LED Preferences to change the endpoint is called.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-03-19:
Thank you for reporting this problem, fixed! Unfortunate that this made it into 0.13, so I'll release 0.13.2 soon.

closed 2022-03-19 by Aircoookie

maujabur, 2022-06-04:
> Thank you for reporting this problem, fixed! Unfortunate that this made it into 0.13, so I'll release 0.13.2 soon.

Hi!
While we don't have 0.13.2, are there any instructions on how to install a version with this fix?

thanks a lot

blazoncek, 2022-06-04:
Compile from source or use binaries provided by @srg74

cbennett6008, 2022-08-15:
I am testing this behavior on the WLED 13.2 that was released yesterday.  Although the release notes seem to indicate this fix was included, I seem to be having the same challenge still. Can anyone else reproduce this behavior?

commit d280e167 (2022-03-19): Fixed `/json/cfg` unable to set busses (fixes #2589)