wled/WLED #2590: Sunrise/set times not calculated correctly

Vic-Y, 2022-03-19:
### What happened?

Sunrise and sunset times are not calculated correctly.

### To Reproduce Bug

Time zone: US-MST/MDT
Latitude N 53.00
Longitude W 113.00
On March 19, 2022, the calculated sunrise is 01:39 and sunset is 13:52

### Expected Behavior

Actual sunrise is 07:39 MDT and sunset is 18:44 MDT

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 (build 2203150)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-03-19:
Thank you for reporting this bug!
I have already fixed it today in master, however the math error leading to incorrect time calculation only manifests itself on March 19th and 20th, so it will also work again without patch in two days time.

Aircoookie, 2022-03-23:
Closing since this bug should not re-occur before March 19 2023, and a fixed release will be available in the short term.

closed 2022-03-23 by Aircoookie