wled/WLED #2591: Ethernet set-up ESP32-ETHERNET-KIT-VE

Roelof626, 2022-03-20:
Hi, I recently tried setting up WLED on a ESP32-ETHERNET-KIT-VE from Espressif. Everything works fine except for the Ethernet connection. I flashed the ESP32-Ethernet bin file and tried all the Ethernet configuration options via the captive portal. But none of them works. I think it has to do with the support for this board. Is this possible? And if not... What else can it be?

Aircoookie, 2022-03-20:
Hi, indeed the pin configuration of that board is not yet supported by WLED, but I can likely add an entry for it!

Roelof626, 2022-03-20:
If isn't to much work, that would be very nice!

Aircoookie, 2022-03-20:
Added! Please confirm whether it works, if you need a binary, it's here: https://github.com/Aircoookie/WLED/suites/5729980419/artifacts/189687712

Roelof626, 2022-03-21:
I flashed the firmware to the board and was installed without any trouble. I set the Ethernet board in the WiFi config to the new one added. and saved the settings. the board connected via ethernet, this was visible via the LAN status LED's. except it's nog getting an IP-address. I checked all router settings and everything is fine in there. I tried restarting the board by switching it off and on. but it forgot all the settings and the wired connection would not even be there anymore.
You have any idea on this?

Aircoookie, 2022-03-22:
Sorry about that. What tool did you use for flashing? I'd recommend installing v0.13.1 with install.wled.me first and afterwards flashing the artifact binary via a wireless OTA update in settings.
It is possible that I got the pin definitions wrong, or that something else in the code needs to be changed that I am not anticipating right now, since your board uses a different PHY (ethernet interface) from all other Ethernet boards supported so far.

Roelof626, 2022-03-24:
I used esptool.py for flashing.
I tried installing via the web installer as you suggested and updated the board via the OTA update.
all went fine until I (again) selected the board type and everything froze. My led are standing still even I had an effect on.
The web interface isn't loading anymore and the app says it is unreachable. It's still connected over WiFi but no IP for the ethernet connection. 
If you don't want to work on it, maybe I can continue the experiment?

martymcbro, 2024-08-23:
This bug still seems to exist.
I connected to the settings page using the built-in AP. When the Config "KIT-VE" is activated, no connection is possible over ethernet (no ip address assigned by DHCP) . The settings page is not accessible anymore over wifi although the wifi connection is still established. Connection over wifi is possible again after restarting the system, however the Ethernet setting is reset to none.

I compiled the latest main branch with debug flags enabled and got the following output over serial connection:

---WLED 0.14.4 2405180 INIT---
esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ5 rev.3, 2 core(s), 240MHz.
FLASH: 4MB, Mode 3 (DOUT), speed 40MHz.
heap 290336
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
Registering usermods ...
heap 290336
Reading config
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
Starting usermod config.
JSON buffer released. (1)
Initializing strip
No busses, init default
PIN ALLOC: Pin 16 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 30) with type 22 and pins 16,0 (itype 21)
Loading custom palettes
Loading custom ledmaps
heap 287012
Usermods setup
heap 287012
initServer
heap 282896
lastReconnectAttempt == 0
No connection configured.
Opening access point 
Init AP interfaces
Loop took 137ms.
Usermods took 0ms.
Strip took 1ms.
Connected AP clients: 1
Not-Found HTTP call:
URI: /connecttest.txt
Captive portal
Not-Found HTTP call:
URI: /connecttest.txt
Captive portal
Not-Found HTTP call:
URI: /connecttest.txt
Captive portal
Not-Found HTTP call:
URI: /connecttest.txt
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
settings resp1
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
Not-Found HTTP call:
URI: /wpad.dat
Captive portal
initE: Attempting ETH config: 7
PIN ALLOC: Pin 21 allocated by 0x81 (129)
PIN ALLOC: Pin 19 allocated by 0x81 (129)
PIN ALLOC: Pin 22 allocated by 0x81 (129)
PIN ALLOC: Pin 25 allocated by 0x81 (129)
PIN ALLOC: Pin 26 allocated by 0x81 (129)
PIN ALLOC: Pin 27 allocated by 0x81 (129)
PIN ALLOC: Pin 23 allocated by 0x81 (129)
PIN ALLOC: Pin 18 allocated by 0x81 (129)
PIN ALLOC: Pin 5 allocated by 0x81 (129)
PIN ALLOC: Pin 0 allocated by 0x81 (129)
Loop took 40ms.
Usermods took 0ms.
Strip took 1ms.
---DEBUG INFO---
Runtime: 30000
Unix time: 30,000
Free heap: 161492
Wifi state: 6
State time: 30001
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 1962
Loop time[ms]: 0/137
UM time[ms]: 0/7
Strip time[ms]: 0/11
Segments: 1 -> 68B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Buffer: 3*30=90B
Local time: 00:01
---DEBUG INFO---
Runtime: 60017
Unix time: 60,017
Free heap: 104672
Wifi state: 6
State time: 30001
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 1419
Loop time[ms]: 1/3
UM time[ms]: 0/1
Strip time[ms]: 0/1

heyitsyang, 2025-05-03:
Any update on this issue?  I have the same problem as described by @martymcbro