wled/WLED #2592: Impossible connect WLED to my ESP32 WROOM 32 (AZ Delivery)

SckyzO, 2022-03-20:
### What happened?

Hello everybody
I have a problem when I try to connect my ESP32 WROMM 32 device (https://www.amazon.fr/gp/product/B071P98VTG).
I download the latest version 0.13.1 (ESP32.bin), I flash my ESP correctly without worries, I have the web interface and I configure my WIFI. But once I connect my led strip (for now 20 led to test). I can't do anything.
I connect the 5v on the 5v, DIN/DO is connected on CMD, ok, but then the GND I connect it on the g16 as indicated in the doc, but I lose the wifi. So I don't understand where the problem comes from.
Thank you for your help

### To Reproduce Bug

N/A

### Expected Behavior

N/A

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-03-20:
Hey, please try the connections as follows:
GND - GND (any of the 4 GND pins really, but closer to USB port is better)
DIN - G2 (make sure pin 2 is set as LED pin in LED settings, it should already be the default)
5V - 5V (**If you want to connect lots of LEDs later, do not power them from the ESP32**)

SckyzO, 2022-03-20:
Wep, I tried... But now indeed, without external power supply, the leds light up... But... Impossible to manage them via Wled. It's very strange, when I touch the different settings on the LEDs in the WLED control panel, I have leds that randomly turn on, turn off, have random colors ... it's very strange. And, when I apply settings all leds poweron in white mode. 
Am I the only one this has happened to?

SckyzO, 2022-03-20:
Whaou perfect ! My DIN was on CMD ... :( 
It's is working perfectly thank ! I close issue

closed 2022-03-20 by SckyzO