wled/WLED #2594: Adding optional DMX input

wystup, 2022-03-23:
In case to avoid problems with WiFi connection in specific places i would suggest to add DMX input option in DMX mod

Allowing DMX mod to run on WLED compilation - it will be great to switch between input or output of DMX data. In this case we can use just one MAX485 IC to switch between two modes by applying adequate voltage on MAX485 (so just one pin will be required to control the mode of DMX flow)  so having two XLR connectors in parallel allow to either send or receive data.

For any of you interested in this modification I can design PCB which will handle this mod.

derpeter, 2022-06-29:
This sounds very interesting. For me it would already be enough to decide on compile / soldering time. My usecase would be to hook one WLED device to a DMX controller and a number of WLED devices to lamps

Aircoookie, 2023-01-02:
Closing in favor of #2824

closed 2023-01-02 by Aircoookie

wystup, 2023-01-03:
Hello,
In case if you will need some help testing this feature I would be happy to
help. Please let me know when source code will be ready to to upload and
test. :)
Best regards,
Pawel Wystup

Imperattor, 2023-02-10:
> Hello,
> In case if you will need some help testing this feature I would be happy to
> help. Please let me know when source code will be ready to to upload and
> test. :)
> Best regards,
> Pawel Wystup
> 
> 
> > Closed #2594 <https://github.com/Aircoookie/WLED/issues/2594> as
> > completed.
> >
> > <https://github.com/Aircoookie/WLED/issues/2594#event-8136545581>, or
> > unsubscribe

Hi, Pawel!
Please, Can you share scheme or diagram how I can wiring physical dmx to esp8266 or other wled board?

Jredbart, 2023-02-12:
Hi @wystup @derpeter @Imperattor,

See another thread that this ticket was closed in favour of. It may be worth directing all questions into the new thread (#2824)

Hope that's not an issue. It just keeps the open issues cleaner.

Many thanks