wled/WLED #2596: ESP32C3 rebooting on sliders interface

TheBattleWolf, 2022-03-24:
### What happened?

As soon as I hit the button "TO THE CONROLS!" the controller crashed and reboots.

### To Reproduce Bug

1. Compile and upload current master branch to an [ESP32C3](https://de.aliexpress.com/item/1005003348936965.html)
2. Connect to the AP
3. Click on "TO THE CONROLS!"
4. Reboot happens

### Expected Behavior

WLED should work :^)

### Install Method

Self-Compiled

### What version of WLED?

master

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4204a014  RA      : 0x4204a068  SP      : 0x3fcb02a0  GP      : 0x3fc8cc00  
TP      : 0x3fc8db64  T0      : 0x40388eb8  T1      : 0xffffffe0  T2      : 0xffffffff  
S0/FP   : 0x3fcb375c  S1      : 0x3fcb4124  A0      : 0x00000001  A1      : 0xf35d0d73  
A2      : 0x3fcb4008  A3      : 0x00000001  A4      : 0x3fc9b6bc  A5      : 0x600c2000  
A6      : 0x3076616b  A7      : 0x71325a45  S2      : 0x3fc9bf54  S3      : 0x3fcb4134  
S4      : 0x3fcb41d0  S5      : 0x00000001  S6      : 0x0000000a  S7      : 0x00000004  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x7342354c  T4      : 0x39203a79  T5      : 0x654b2d74  T6      : 0x656b636f  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xf35d0d82  
MHARTID : 0x00000000  

Stack memory:
3fcb02a0: 0x00000000 0x00000000 0x00000000 0x0404c808 0x0000000a 0x00000001 0x3fcb4158 0x00000002
3fcb02c0: 0x3fcb46b9 0x00000002 0x3fcb4124 0x4204a0ea 0x3fcb46b9 0x00000002 0x3fcb4158 0x420507e0
3fcb02e0: 0x0000000a 0x00000001 0x3fcb4158 0x00000002 0x3fcb46b9 0x00000002 0x3fcb4124 0x4204a1e8
3fcb0300: 0x00000000 0x00000000 0x00000000 0x00000004 0x00000002 0x00000005 0x3fc9b000 0x3fc9b2d4
3fcb0320: 0x00000002 0x3fcb40bc 0x3fcb40a8 0x00000001 0x00000000 0x3fcb44bc 0x3fcb4050 0x4204a3b8
3fcb0340: 0x00000000 0x3fcb44bc 0x3fcb4050 0x420422b6 0x3fcb4124 0x3fcb4050 0x3fcb450a 0x000001b1
3fcb0360: 0x00000002 0x00000005 0x3fc9b000 0x3fc9b2d4 0x00000001 0x3fcb4050 0x3fcb3a00 0x4204239a
3fcb0380: 0x00000001 0x3fcb4050 0x3fcb3a00 0x42042a4a 0x00000000 0x3fcb3a00 0x00000000 0x00000000
3fcb03a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb03c0: 0x00000000 0x00000000 0x00000000 0x40388e60 0x00000000 0x00000000 0x00000000 0x00000000
3fcb03e0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xffffff02 0x000000ff 0x27146414 0x01010107
3fcb0400: 0x00000001 0x00000000 0x35d1cfdc 0x00000000 0x3f15c000 0x3f800000 0x3fc00000 0x4b000000
3fcb0420: 0xcb000000 0x000000ff 0x00000000 0x8800759c 0x7e37e43c 0x7149f2ca 0x00000000 0x00000000
3fcb0440: 0xfff00000 0x00000000 0x7ff00000 0x60000000 0x3ff71547 0xf85ddf44 0x3e54ae0b 0x00000000
3fcb0460: 0x3fd00000 0x55555555 0x3fd55555 0x652b82fe 0x3ff71547 0x00000000 0x43400000 0x4a454eef
3fcb0480: 0x3fca7e28 0x93c9db65 0x3fcd864a 0xa91d4101 0x3fd17460 0x518f264d 0x3fd55555 0xdb6fabff
3fcb04a0: 0x3fdb6db6 0x33333303 0x3fe33333 0xe0000000 0x3feec709 0xdc3a03fd 0x3feec709 0x145b01f5
3fcb04c0: 0xbe3e2fe0 0x652b82fe 0x3c971547 0xc2f8f359 0x01a56e1f 0x00000000 0x3fe62e43 0xfefa39ef
3fcb04e0: 0x3fe62e42 0x0ca86c39 0xbe205c61 0x72bea4d0 0x3e663769 0xc5d26bf1 0x3ebbbd41 0xaf25de2c
3fcb0500: 0x3f11566a 0x16bebd93 0x3f66c16c 0x5555553e 0x3fc55555 0xcc000000 0x4c000000 0x3f317180
3fcb0520: 0x3717f7d1 0x3eaaaaab 0x3e178897 0x3e3a3325 0x3e924925 0x3f2aaaab 0x3e1cd04f 0x3e638e29
3fcb0540: 0x3ecccccd 0x3fb8aa00 0x36eca570 0x3fb8aa3b 0x4b800000 0x3e53f142 0x3e6c3255 0x3e8ba305
3fcb0560: 0x3edb6db7 0x3f19999a 0x3f763800 0x3f76384f 0x369dc3a0 0x3338aa3c 0x0da24260 0x3f317200
3fcb0580: 0x3f317218 0x35bfbe8c 0x3331bb4c 0x35ddea0e 0x388ab355 0x3b360b61 0x3e2aaaab 0x00000000
3fcb05a0: 0x7ff80000 0x00000000 0x43500000 0x8800759c 0xfe37e43c 0xc2f8f359 0x81a56e1f 0x00000000
3fcb05c0: 0x3c900000 0x7fc00000 0xf149f2ca 0x8da24260 0x33000000 0x7f7fffff 0x00000000 0xffc00000
3fcb05e0: 0x41dfffff 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000 0x94a03595 0x3fcfffff 0x00000000
3fcb0600: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fcb0620: 0x401c0000 0x00000000 0x40140000 0x00000018 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
3fcb0640: 0x3fcb0190 0x3fc9bf54 0x3fc97744 0x3fc97744 0x3fcb0640 0x3fc9773c 0x00000016 0x3fcae590
3fcb0660: 0x3fcae590 0x3fcb0640 0x00000000 0x00000003 0x3fcae63c 0x6e797361 0x63745f63 0x83a00070
3fcb0680: 0x009fbaa2 0x00000000 0x3fcb0630 0x00000003 0x00000000 0x3fcb49d0 0x4205cbd0 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382882
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x48
load:0x403ce000,len:0x648
load:0x403d0000,len:0x207c
SHA-256 comparison failed:
Calculated: 6ebe8f1675f66ce3902e819fac8d71b0a3fe08ee9da84924d961cc49adb110a3
Expected: 2f36a2f70ce415df42057ab4d5b3a85d6f5cad0222a0fae3dbac51e322c4f2c6
Attempting to boot anyway...
entry 0x403ce000
Ada
```

### Anything else?

If i can provide any further help/information let me know :-)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

TheBattleWolf, 2022-03-24:
Okay, I've done like you said, but it doesn't seem to change anything.
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4204a014  RA      : 0x4204a068  SP      : 0x3fcafe00  GP      : 0x3fc8cc00  
TP      : 0x3fc8d6c4  T0      : 0x40388eb8  T1      : 0xffffffe0  T2      : 0xffffffff  
S0/FP   : 0x3fcb4194  S1      : 0x3fcb339c  A0      : 0x00000001  A1      : 0xf35d0d33  
A2      : 0x3fcb40f8  A3      : 0x00000001  A4      : 0x3fc9b6bc  A5      : 0x600c2000  
A6      : 0x552b5545  A7      : 0x78444e45  S2      : 0x3fc9bf54  S3      : 0x3fcb33ac  
S4      : 0x3fcb3448  S5      : 0x00000001  S6      : 0x0000000a  S7      : 0x00000004  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x69795638  T4      : 0x4f203a79  T5      : 0x654b2d74  T6      : 0x656b636f  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xf35d0d42  
MHARTID : 0x00000000  

Stack memory:
3fcafe00: 0x00000000 0x00000000 0x00000000 0x0404c808 0x0000000a 0x00000001 0x3fcb33d0 0x00000002
3fcafe20: 0x3fcb36e1 0x00000002 0x3fcb339c 0x4204a0ea 0x3fcb36e1 0x00000002 0x3fcb33d0 0x420507e0
3fcafe40: 0x0000000a 0x00000001 0x3fcb33d0 0x00000002 0x3fcb36e1 0x00000002 0x3fcb339c 0x4204a1e8
3fcafe60: 0x00000000 0x00000000 0x00000000 0x00000004 0x00000002 0x00000005 0x3fc9b000 0x3fc9b2d4
3fcafe80: 0x00000002 0x3fcb3334 0x3fcb3320 0x00000001 0x00000000 0x3fcb34e4 0x3fcb32c8 0x4204a3b8
3fcafea0: 0x00000000 0x3fcb34e4 0x3fcb32c8 0x420422b6 0x3fcb339c 0x3fcb32c8 0x3fcb3532 0x000001b1
3fcafec0: 0x00000002 0x00000005 0x3fc9b000 0x3fc9b2d4 0x00000001 0x3fcb32c8 0x3fcb416c 0x4204239a
3fcafee0: 0x00000001 0x3fcb32c8 0x3fcb416c 0x42042a4a 0x00000000 0x3fcb416c 0x00000000 0x00000000
3fcaff00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaff20: 0x00000000 0x00000000 0x00000000 0x40388e60 0x00000000 0x00000000 0x00000000 0x00000000
3fcaff40: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xffffff02 0x000000ff 0x27146414 0x01010107
3fcaff60: 0x00000001 0x00000000 0x35d1cfdc 0x00000000 0x3f15c000 0x3f800000 0x3fc00000 0x4b000000
3fcaff80: 0xcb000000 0x000000ff 0x00000000 0x8800759c 0x7e37e43c 0x7149f2ca 0x00000000 0x00000000
3fcaffa0: 0xfff00000 0x00000000 0x7ff00000 0x60000000 0x3ff71547 0xf85ddf44 0x3e54ae0b 0x00000000
3fcaffc0: 0x3fd00000 0x55555555 0x3fd55555 0x652b82fe 0x3ff71547 0x00000000 0x43400000 0x4a454eef
3fcaffe0: 0x3fca7e28 0x93c9db65 0x3fcd864a 0xa91d4101 0x3fd17460 0x518f264d 0x3fd55555 0xdb6fabff
3fcb0000: 0x3fdb6db6 0x33333303 0x3fe33333 0xe0000000 0x3feec709 0xdc3a03fd 0x3feec709 0x145b01f5
3fcb0020: 0xbe3e2fe0 0x652b82fe 0x3c971547 0xc2f8f359 0x01a56e1f 0x00000000 0x3fe62e43 0xfefa39ef
3fcb0040: 0x3fe62e42 0x0ca86c39 0xbe205c61 0x72bea4d0 0x3e663769 0xc5d26bf1 0x3ebbbd41 0xaf25de2c
3fcb0060: 0x3f11566a 0x16bebd93 0x3f66c16c 0x5555553e 0x3fc55555 0xcc000000 0x4c000000 0x3f317180
3fcb0080: 0x3717f7d1 0x3eaaaaab 0x3e178897 0x3e3a3325 0x3e924925 0x3f2aaaab 0x3e1cd04f 0x3e638e29
3fcb00a0: 0x3ecccccd 0x3fb8aa00 0x36eca570 0x3fb8aa3b 0x4b800000 0x3e53f142 0x3e6c3255 0x3e8ba305
3fcb00c0: 0x3edb6db7 0x3f19999a 0x3f763800 0x3f76384f 0x369dc3a0 0x3338aa3c 0x0da24260 0x3f317200
3fcb00e0: 0x3f317218 0x35bfbe8c 0x3331bb4c 0x35ddea0e 0x388ab355 0x3b360b61 0x3e2aaaab 0x00000000
3fcb0100: 0x7ff80000 0x00000000 0x43500000 0x8800759c 0xfe37e43c 0xc2f8f359 0x81a56e1f 0x00000000
3fcb0120: 0x3c900000 0x7fc00000 0xf149f2ca 0x8da24260 0x33000000 0x7f7fffff 0x00000000 0xffc00000
3fcb0140: 0x41dfffff 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000 0x94a03595 0x3fcfffff 0x00000000
3fcb0160: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fcb0180: 0x401c0000 0x00000000 0x40140000 0x00000018 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb01a0: 0x00000154 0x3fcafd50 0x3fc9bf54 0x3fc97744 0x3fc97744 0x3fcb01a4 0x3fc9773c 0x00000016
3fcb01c0: 0x3fcae0f4 0x3fcae0f4 0x3fcb01a4 0x00000000 0x00000003 0x3fcae1a0 0x6e797361 0x63745f63
3fcb01e0: 0x33c90070 0x006c9a83 0x00000000 0x3fcb0190 0x00000003 0x00000000 0x3fcb37e4 0x4205cbd0

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382882
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x48
load:0x403ce000,len:0x648
load:0x403d0000,len:0x207c
SHA-256 comparison failed:
Calculated: 6ebe8f1675f66ce3902e819fac8d71b0a3fe08ee9da84924d961cc49adb110a3
Expected: 2f36a2f70ce415df42057ab4d5b3a85d6f5cad0222a0fae3dbac51e322c4f2c6
Attempting to boot anyway...
entry 0x403ce000
Ada
```
Here is the git diff to make sure:
```
@@ -12,12 +12,13 @@
 ; default_envs = travis_esp8266, travis_esp32

 # Release binaries
-default_envs = nodemcuv2, esp8266_2m, esp01_1m_full, esp32dev, esp32_eth, esp32s2_saola, esp32c3
+;default_envs = nodemcuv2, esp8266_2m, esp01_1m_full, esp32dev, esp32_eth, esp32s2_saola, esp32c3

 # Build everything
 ; default_envs = esp32dev, esp8285_4CH_MagicHome, codm-controller-0.6-rev2, codm-controller-0.6, esp32s2_saola, d1_mini_5CH_Shojo_PCB, d1_mini, sp501e, travis_esp8266, travis_esp32, nodemcuv2, esp32_eth, anavi_miracle_controller, esp07, esp01_1m_full, m5atom, h803wf, d1_mini_ota, heltec_wifi_kit_8, esp8285_H801, d1_mini_debug, wemos_shield_esp32, elekstube_ips

 # Single binaries (uncomment your board)
+default_envs = esp32c3
 ; default_envs = elekstube_ips
 ; default_envs = nodemcuv2
 ; default_envs = esp8266_2m
@@ -359,6 +360,7 @@ board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
 upload_speed = 460800
 build_unflags = ${common.build_unflags}
 lib_deps = ${esp32c3.lib_deps}
+monitor_filters = esp32_exception_decoder

 [env:esp8285_4CH_MagicHome]
 board = esp8285
```

fallspectrum, 2022-05-07:
Using latest on master, commit bef9c68f8152cad463fc8dbf6c143d6c81ea3434

On ESP32-C3-DevKitM-1, with the following platform_override.ini (comments removed):

```
[platformio]
default_envs = esp32c3

[env:esp32c3]
board = esp32-c3-devkitm-1
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.2/platform-tasmota-espressif32-2.0.2.zip
platform_packages =
framework = arduino
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
upload_speed = 460800
lib_deps = ${esp32c3.lib_deps}
monitor_filters = esp32_exception_decoder
build_unflags = ${common.build_unflags}
build_flags = -g
  -DARDUINO_ARCH_ESP32
  -DARDUINO_ARCH_ESP32C3
  -DCONFIG_IDF_TARGET_ESP32C3
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -DCO
; ******** non-stand build flags defined below
   -D WLED_USE_MY_CONFIG
   -D USE_APA102
   -D CLKPIN=6
   -D DATAPIN=7
   -D SERVERNAME="\"WLED\""
   -D DEFAULT_LED_COUNT=10
   -D ABL_MILLIAMPS_DEFAULT=850
```

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42049e84  RA      : 0x42049ed8  SP      : 0x3fcaff10  GP      : 0x3fc8cc00  
TP      : 0x3fc8d6b0  T0      : 0x40388e58  T1      : 0xffffffe0  T2      : 0xffffffff  
S0/FP   : 0x3fcb3410  S1      : 0x3fcb46a4  A0      : 0x00000001  A1      : 0x781d7c28  
A2      : 0x3fcb4598  A3      : 0x00000001  A4      : 0x3fc9b6dc  A5      : 0x600c2000  
A6      : 0x45574d48  A7      : 0x4d49374e  S2      : 0x3fc9bf74  S3      : 0x3fcb46b4  
S4      : 0x3fcb4750  S5      : 0x00000001  S6      : 0x0000000a  S7      : 0x00000004  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x49646876  T4      : 0x66203a79  T5      : 0x654b2d74  T6      : 0x656b636f  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x781d7c37  
MHARTID : 0x00000000  

Stack memory:
3fcaff10: 0x00000000 0x00000000 0x00000000 0x0604c678 0x0000000a 0x00000001 0x3fcb46d8 0x00000002
3fcaff30: 0x3fcb52d7 0x00000002 0x3fcb46a4 0x42049f5a 0x3fcb52d7 0x00000002 0x3fcb46d8 0x42050650
3fcaff50: 0x0000000a 0x00000001 0x3fcb46d8 0x00000002 0x3fcb52d7 0x00000002 0x3fcb46a4 0x4204a058
3fcaff70: 0x3fcb42bc 0x3c0e2000 0x3fcb46a4 0x42049e1e 0x00000001 0x3fc94608 0x42047eea 0x42048010
3fcaff90: 0x00000002 0x3fcb4328 0x3fcb4314 0x00000001 0x00000000 0x3fcb50a4 0x3fcb42bc 0x4204a228
3fcaffb0: 0x00000000 0x3fcb50a4 0x3fcb42bc 0x42042132 0x3fcb46a4 0x3fcb42bc 0x3fcb50f2 0x000001e7
3fcaffd0: 0x00000002 0x00000005 0x3fc9b000 0x3fc9b2f4 0x00000001 0x3fcb42bc 0x3fcb4390 0x42042216
3fcafff0: 0x00000001 0x3fcb42bc 0x3fcb4390 0x420428ba 0x00000000 0x3fcb4390 0x00000000 0x00000000
3fcb0010: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb0030: 0x00000000 0x00000000 0x00000000 0x40388e00 0x00000000 0x00000000 0x00000000 0x00000000
3fcb0050: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xffffff02 0x000000ff 0x27146414 0x01010107
3fcb0070: 0x00000001 0x00000000 0x35d1cfdc 0x00000000 0x3f15c000 0x3f800000 0x3fc00000 0x4b000000
3fcb0090: 0xcb000000 0x000000ff 0x8800759c 0x7e37e43c 0x7149f2ca 0x00000000 0x00000000 0xfff00000
3fcb00b0: 0x00000000 0x7ff00000 0x60000000 0x3ff71547 0xf85ddf44 0x3e54ae0b 0x00000000 0x3fd00000
3fcb00d0: 0x55555555 0x3fd55555 0x652b82fe 0x3ff71547 0x00000000 0x43400000 0x4a454eef 0x3fca7e28
3fcb00f0: 0x93c9db65 0x3fcd864a 0xa91d4101 0x3fd17460 0x518f264d 0x3fd55555 0xdb6fabff 0x3fdb6db6
3fcb0110: 0x33333303 0x3fe33333 0xe0000000 0x3feec709 0xdc3a03fd 0x3feec709 0x145b01f5 0xbe3e2fe0
3fcb0130: 0x652b82fe 0x3c971547 0xc2f8f359 0x01a56e1f 0x00000000 0x3fe62e43 0xfefa39ef 0x3fe62e42
3fcb0150: 0x0ca86c39 0xbe205c61 0x72bea4d0 0x3e663769 0xc5d26bf1 0x3ebbbd41 0xaf25de2c 0x3f11566a
3fcb0170: 0x16bebd93 0x3f66c16c 0x5555553e 0x3fc55555 0xcc000000 0x4c000000 0x3f317180 0x3717f7d1
3fcb0190: 0x3eaaaaab 0x3e178897 0x3e3a3325 0x3e924925 0x3f2aaaab 0x3e1cd04f 0x3e638e29 0x3ecccccd
3fcb01b0: 0x3fb8aa00 0x36eca570 0x3fb8aa3b 0x4b800000 0x3e53f142 0x3e6c3255 0x3e8ba305 0x3edb6db7
3fcb01d0: 0x3f19999a 0x3f763800 0x3f76384f 0x369dc3a0 0x3338aa3c 0x0da24260 0x3f317200 0x3f317218
3fcb01f0: 0x35bfbe8c 0x3331bb4c 0x35ddea0e 0x388ab355 0x3b360b61 0x3e2aaaab 0x00000000 0x7ff80000
3fcb0210: 0x00000000 0x43500000 0x8800759c 0xfe37e43c 0xc2f8f359 0x81a56e1f 0x00000000 0x3c900000
3fcb0230: 0x7fc00000 0xf149f2ca 0x8da24260 0x33000000 0x7f7fffff 0x00000000 0xffc00000 0x41dfffff
3fcb0250: 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000 0x94a03595 0x3fcfffff 0x00000000 0x3ff80000
3fcb0270: 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000 0x401c0000
3fcb0290: 0x00000000 0x40140000 0x00000010 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb02b0: 0x00000154 0x3fcafe60 0x3fc9bf74 0x3fc97764 0x3fc97764 0x3fcb02b4 0x3fc9775c 0x00000016
3fcb02d0: 0x3fcae204 0x3fcae204 0x3fcb02b4 0x00000000 0x00000003 0x3fcae2b0 0x6e797361 0x63745f63
3fcb02f0: 0xc5860070 0x0009434c 0x00000000 0x3fcb02a0 0x00000003 0x00000000 0x3fcb4684 0x4205ca40

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382822
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x48
load:0x403ce000,len:0x648
load:0x403d0000,len:0x207c
SHA-256 comparison failed:
Calculated: 6ebe8f1675f66ce3902e819fac8d71b0a3fe08ee9da84924d961cc49adb110a3
Expected: 2f36a2f70ce415df42057ab4d5b3a85d6f5cad0222a0fae3dbac51e322c4f2c6
Attempting to boot anyway...
entry 0x403ce000
```

fallspectrum, 2022-05-07:
Also encountering issues with decoding or getting a stack trace with the given stack dump: 

https://github.com/me-no-dev/EspExceptionDecoder/issues/80
https://github.com/platformio/platform-espressif32/pull/612 (stale WIP draft PR)

PW999, 2022-08-15:
I managed to fix the reboots on the slider interface bug and stacktraces are working too for me now, I've put some more info here: https://github.com/Aircoookie/WLED/issues/1940#issuecomment-1214424062

It's still not 100% stable though, I still get random reboots with the same "load access fault" error, sometimes after one click, sometimes after randomly bashing on different effects for a minute.

blazoncek, 2022-10-03:
Building 0_14 branch with:
```
platform = espressif32@5.1.1
framework = arduino
board = esp32-c3-devkitm-1
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
upload_port = /dev/cu.usbmodem1A12301
monitor_port = /dev/cu.usbmodem1A12301
build_flags = ${common.build_flags} ${esp32c3.build_flags} #-D WLED_RELEASE_NAME=ESP32-C3
  -D WLED_WATCHDOG_TIMEOUT=0
  -D LEDPIN=8
```

produces working instance without issues (on ESP32-C3-DevKitM-1 board).

blazoncek, 2022-11-18:
ESP32-C3 has been working solid for over a month of continuous testing.
Closing.

closed 2022-11-18 by blazoncek