wled/WLED #2598: ESP32 crash when activating "Peek"

dasadi, 2022-03-24:
### What happened?

When activating the "peek" feature in the ui the ESP32 crashes immediately.
ESP32 D1 mini on Quinled-dig-quad in AP mode (not connected to WiFi) with 300x SK6812 LEDs
White Balance correction is activated, but crash happens also without WB correction.
Before crashing all LEDs turn completely white for a split second before turning off.
Installed on a completely new ESP32  using ESP Web Tools at [https://install.wled.me/](url) 

Everything else working as expected, 
I'm happy to provide logs if needed (Do I need to reflash with a debug version to obtain logs?)

### To Reproduce Bug

While the LEDs are turned on, clicking the "peek"-Button in UI produces the crash

### Expected Behavior

Peek should display a representation of currently shown colors on the strip

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 (build 2203150)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dasadi, 2022-03-25:
Definitely same issue as #2586, current master indeed fixes the issue; And it's my bad, I'm currently running an uneven number of LEDs :\ 
Thank you for the heads up, I will look for open issues more closely next time.

closed 2022-03-25 by dasadi

Aircoookie, 2022-03-25:
No worries, thanks for the quick confirmation and glad that that was it! :)