wled/WLED #2600: WLED 13.1 on ESP01 - Network lost...GUI not resposible

Diddlik, 2022-03-25:
### What happened?

I have installed the current version WLED 13.1 ESP01 on the ESP01. WLED AP is set up, but when I try to log in, I get an error message: Access denied or simply reboot. After several attempts, it works at some point, but then the error messages as shown in the picture appear. If I do nothing and simply start ping, it is not stable either.

[image]

### To Reproduce Bug

Just install on ESP01 ant try to use it

### Expected Behavior

I should be able to login and setup the controller. 

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Reply from 192.168.1.225: bytes=32 time=24ms TTL=128
Reply from 192.168.1.225: bytes=32 time=19ms TTL=128
Reply from 192.168.1.225: bytes=32 time=18ms TTL=128
Reply from 192.168.1.225: bytes=32 time=21ms TTL=128
Reply from 192.168.1.225: bytes=32 time=49ms TTL=128
Reply from 192.168.1.225: bytes=32 time=30ms TTL=128
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.16: Destination host unreachable.
Reply from 192.168.1.225: bytes=32 time=1920ms TTL=128
Reply from 192.168.1.225: bytes=32 time=31ms TTL=128
Reply from 192.168.1.225: bytes=32 time=28ms TTL=128
Reply from 192.168.1.225: bytes=32 time=74ms TTL=128
Request timed out.
Request timed out.
Reply from 192.168.1.225: bytes=32 time=45ms TTL=128
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.1.225: bytes=32 time=99ms TTL=128
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
```
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DwaineGarden, 2022-03-28:
I'm getting the same thing with ESP32 build on 13.1.   It happens about every 10 seconds with device rebooting.

blazoncek, 2022-03-28:
Please try experimental binary from @srg74 Wemos shield repository.

DwaineGarden, 2022-03-28:
Confirming that the experimental build the problem is now resolved.  No more reboots and the web interface stays online continuously.

birdofhermes, 2022-03-31:
I have the same issue as the original poster with the same ESP8266 controller type. Ping is not constant. Sometimes it has a flow of 1k successfull pings and sometimes it changes status after 2-3 ping attempts.
Tested with 13.0B6, 13.1 and the above linked experimental same issues.
Created a Telegram notifyer in NodeRed and it does all the pings for me. Here is tonights transcript.
TGBot, [2022. 03. 30. 23:47]
WLED went offline! After: 3 of online pings.

TGBot, [2022. 03. 30. 23:48]
WLED is back online! After: 3 of offline pings.

TGBot, [2022. 03. 31. 2:18]
WLED went offline! After: 2215 of online pings.

TGBot, [2022. 03. 31. 2:18]
WLED is back online! After: 3 of offline pings.

PS: Offline pings mean the timeouts.

blazoncek, 2022-03-31:
You may have network issues or your ESPs may be faulty.
Most likely not a WLED problem.

Please check which ESP core your particular WLED build is using and then try to compile with different cores and see which works best for your HW.

Diddlik, 2022-04-02:
> Please try experimental binary from @srg74 Wemos shield repository.
> 
> https://github.com/srg74/WLED-wemos-shield/tree/master/resources/experimental/Firmware

Will be the the fix available in the next wled release?

blazoncek, 2022-04-03:
Does it work? Some of the changes were already submitted to main WLED but it may take a while.

Diddlik, 2022-04-03:
> Does it work? Some of the changes were already submitted to main WLED but it may take a while.

I'm only home tonight.  I will test it

Diddlik, 2022-04-04:
I used this one esp01_1m_full_shield.bin, after flash I am not able to access WLED-AP, I get "Access Denied" or Time Out

lplassman, 2022-04-08:
I am getting the same problem on all my ESP8266 and ESP32 devices. They used to be rock solid on version 12 of WLED, now getting 'connection to light failed' errors in the GUI and they are unresponsive.

blazoncek, 2022-04-08:
Try factory reset or erase flash and flash again.

lplassman, 2022-04-08:
> Try factory reset or erase flash and flash again.

I just re-flashed and the issue still persists. It seems to be more apparent on my ESP8266 vs. ESP32 devices.

olialb, 2022-04-09:
Same issue here here with ESP8266 (WEMOS D1) board and 13.1. I don't know why it god worse agian. With 13 beta it seamd better. I programmed a script now and use another esp with Tasmota to reset the WLED ESP when coneection is lost. This morning it is reset WLED every 15min... Really annoying I must say. And I love this project...

PrzemyslawSarnacki, 2022-04-11:
Same issue here with 13.1 version and ESP-01S. It was basically non usable. Is there any workaround for now or we have to wait?

blazoncek, 2022-04-12:
I have about 20 ESP01 (and ESP01S) installations and all of them are working flawlessly.
I am not saying you do not have issues but WLED 0.13.1 is stable and working correctly if applied to ESP correctly.

Please make sure you download correct binary and use esptool to erase flash first then upload binary, disconnecting ESP from power (PC) between erasing flash and flashing with binary file.

PrzemyslawSarnacki, 2022-04-15:
I'm not sure but I think this might be an issue with power because without connecting to load it seems to be working fine but under the load (I mean when connecting led strip) it crashes

olialb, 2022-04-15:
@PrzemyslawSarnacki  I am also investigating in same direction. It seams only unstable when I plug my Wemos D1 in the circuit with the LED drivers and the Relay to switch the LEDs off. May be one of the two cirquits drains to much power from a D1 port even when the LEDs are off? Stange...

blazoncek, 2022-04-15:
Are you using properly designed ESP shield/controller? If not, use one.
IMO two of the finest are @srg74 Wemos and ESP01 [shields](https://www.tindie.com/stores/serg74/) and @intermittech [Dig](https://quinled.info/) controllers.

olialb, 2022-04-16:
@blazoncek Thanks for the hint. But WLED is also DIY project or. No challange to buy a ready made shield :-) Actually I made good expereience with self made shields for Tasmota. I hope I will get mine for WLED also running. Its really simple. And I appologize if I blamed the software... As I stated before nice project and thanks to all the contributors!

blazoncek, 2022-04-16:
You still don't get it. 😄 Both of those are also available as DIY open source controllers.

PrzemyslawSarnacki, 2022-04-16:
I think that the issue is not not using shield but buck converter (mini360). I have built similar project with 5V power supply, wemos d1 mini and WS2182B Led Strip and it works fine most of the time.  This time I have this 12V PWM led strip with  additional circuit (3 mosfets and some resistors) and converting voltage with this mini360 to 3.3V seems to cause this issue but I'm not really sure

blazoncek, 2022-04-16:
It is all about stable power supply to ESP. I give up.

olialb, 2022-04-16:
I don't think that its in my case Power supply issue... Its also unstable when no LED is on.

olialb, 2022-04-17:
I rebuild my circuit. This time I turned the Wemos D1 arround that ESP and Antenna is outside. On first view it more stable... I will see the next days.

poelzi, 2022-04-28:
I think that might be the same bug as #2551
For me, stability decreased somewhere between 0.11 and 0.13 but is hard to pinpoint. Since I have this with all wled devices on a variaty of power supplies and different models, I think there is some underlaying reason.
I'm using the home-assistant integration.
Maybe some network packet or wlan property might affect this problem. I will try if the problem happens with esp32eth on lan as well.
For me, it looks like a hard lockup since the button is also not working if master is off and I press the user button.

rivelta, 2022-05-06:
i have same problem with wemos d1 R2 , its happens after device reboot then ap ssid gone and led its not responding.

poelzi, 2022-05-08:
@rivelta does the button still work ?

DwaineGarden, 2022-09-21:
My issue has been resolved with the newer builds.   Thanks.  Please close.

closed 2022-09-23 by Aircoookie