wled/WLED #2603: DST is applied to time, but not to sunrise/sunset

carelinus, 2022-03-27:
It seems that the offset for summer time is correctly applied in itself. However, it is not applied to the sunrise and sunset times. Thus, the sunset event takes place one hour too early during summer time.

Aircoookie, 2022-03-27:
Thank you for the report!
The offset is dynamically applied to the sunrise and sunset times (which are calculated in UTC), however apparently not correctly.
For my location WLED calculates a Sunrise time of 07:16 and a Sunset time of 18:53, the true times are 07:16 and 19:53, so it indeed seems that Sunset is an hour early, but sunrise is correct, which is strange since they use the same code.
Will investigate.

blazoncek, 2022-03-27:
IMO the problem lies in sunrise/sunset time being calculated only once per day at midnight.

Aircoookie, 2022-03-27:
Don't think so, since it is still wrong when rebooting after the DST change.
The time of calculation shouldn't matter either, since all `toLocal()` does/should do is convert a UTC time to the corresponding local time, irrespective of the current time where calculation runs.

Aircoookie, 2022-03-28:
It is correct again today (the day after DST change in Europe)

I think it may be due to getSunriseTimeUTC returning negative values for minUTC sometimes, which do correctly wrap around, but may make the timezone lib think it's the previous day...

closed 2022-04-01 by Aircoookie

carelinus, 2022-04-01:
Wow, quickly resolved. Nice, thank you!

commit e362b3b6 (2022-04-01): Fixed sunset time off by an hour on DST change day (fixes #2603)