wled/WLED #2605: last change/input source

nellydocs, 2022-03-29:
I just setup a 600 led strip with sk6812rgbcw's. I also setup a playlist, a Time-controlled preset, Alexa and other features WLED has to offer (which are all AWESOME!!!!). But now i have the problem that my lights sometimes change into different presets or brightness levels. And i cant trace back where the changes are coming from.

A clear and concise description of what you want to happen:
To have a debug toggle in the  options, that prints stuff like,
who changed what, at what time to what ip for example in a json like this
```
{
  "changeRequestFrom" : "Alexa || esp32 || ....... etc",
  "changeRequestTo"     : "192.168.0.246 ",
  "changeObject"           :  
                            [
                              "brightnessFrom" :  "20",
                              "brightnessTo"      : "100",
                              "effectFrom"         : "Android",
                              "effectTo"              : "Solid",
                            ]
}
```

if there are already is such a debug option please let me know ♥

Thanks
Nelly

blazoncek, 2022-03-29:
That info is not maintained within WLED.
Use Wireshark or similar tool to see what's happening on network.

nellydocs, 2022-03-29:
oh yeah i didnt think of this.
guess we can close this

thanks

closed 2022-03-29 by nellydocs

blazoncek, 2022-03-29:
If you can compile yourself, you can still add a bunch of `DEBUG_PRINTLN(...)`at various places and monitor serial output.
You'll need to find the relevant places yourself, though.