wled/WLED #2612: esp8266 settings lost after reboot

sebPomme, 2022-04-01:
### What happened?

Hello, 

After an upgrade (from v0.12.0 to v.0.13.1). I can't keep my settings configured.

### To Reproduce Bug

1 - flash my micro-controler
2 - Reboot
3 - Connect to the hostpost
4 - Set my home wifi settings
5 - Start to edit the configuration
6 - Reboot, and there everything is lost and I have to go to step 3

### Expected Behavior

I expected my all my settings are saved and persist across reboot (like previously).

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1/2203150

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
pomme@requin ~ % esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:cf:1a:cc
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...
pomme@requin ~ %
pomme@requin ~ % esptool.py write_flash 0x0 Downloads/WLED_0.13.1_ESP8266.bin
esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:cf:1a:cc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0320
Compressed 747008 bytes to 505270...
Wrote 747008 bytes (505270 compressed) at 0x00000000 in 44.5 seconds (effective 134.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### Anything else?

For information I have two setups.
One on v0.12.0 / 2104020 still work fine.
The other one I just try to upgrade.

I try to install the firmware through wled.me and thought the cli, same result. 

If you have any inputs / debug tips / knowledge , you are welcome. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-04-01:
Your unit only has 1MB of flash memory, please use the binary ending in `ESP01.bin`

sebPomme, 2022-04-01:
It's works fine on my two nodes, thanks for the quick response. And next time I will read "harder"

closed 2022-04-01 by sebPomme