wled/WLED #2614: Refresh available devices when connected WiFi ssid is different.

pandaboy6621, 2022-04-02:
**Is your feature request related to a problem? Please describe.**
It's always annoying to go between multiple houses with WLED devices. When I am at one location I don't want to see all my devices from my other house cluttering my list because they are unavailable, seeing that I am no longer on that network.
**Describe the solution you'd like**
I would like the WLED app to only show the devices that are currently attached to the network that I am on. If a device is on a different ssid from the one I am on I wanted to be hidden. Maybe there should be an view mode option. 1. To show all devices 2. To show all online local devices and 3. To show all local devices online or offline. All of which reorder the list into a custom order or to have the offline devices automatically go to the bottom of the list.

blazoncek, 2022-04-03:
I would also like to add that sorting/rearranging would be very useful.

Aircoookie, 2022-04-03:
The app has no access to the currently connected SSID on newer versions of Android and (I think) iOS.
Still I agree it would be nice to be able to reorder devices and move unavailable ones to the bottom automatically.

blazoncek, 2022-04-03:
Perhaps adding "Groups" or "Locations" could be an option?
You could automatically group them by IP subnet if SSID is unavailable (there are restrictions in place).

Aircoookie, 2022-04-03:
IP subnet is a great idea, at least if the different networks use separate ranges.
Groups/locations would also be great.
IMO https://play.google.com/store/apps/details?id=ca.cgagnier.wlednativeandroid is better than my old app :) Maybe it could be added there as well

blazoncek, 2022-11-18:
@Moustachauve have you seen this discussion?

Moustachauve, 2022-11-21:
Thanks for pointing out this discussion @blazoncek. Having an option to automatically hide offline device sounds great!

Moustachauve, 2022-12-01:
The current version of my app now support something very similar to this.
There is a setting to show offline devices at the bottom of the list

Also with Automatic discovery turned on (it is on by default), devices are identified by mac address so that when a device changes IP address, it is detected and updated in the app.

softhack007, 2023-10-07:
Closing, as the request cannot be implemented in the WLED core. 
The [mobile app has it's own repo](https://github.com/Aircoookie/WLED-App/issues) on github, please open a feature request there. Or try the new app from @Moustachauve :-).

closed 2023-10-07 by softhack007