wled/WLED #2618: Receiving adalight over serial crashes ESP 8266 and 32

apophisnow, 2022-04-04:
### What happened?

Adalight over serial causes boards to crash.

### To Reproduce Bug

To replicate, use LedFX to send adalight data over a serial connection to ESP 8266 or ESP 32 board. Board will crash and reboot after a short period of time. Baud rate has no impact. Multiple boards were tried.

### Expected Behavior

Serial data should not cause board crashes.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Anything else?

Using https://github.com/dmadison/Adalight-FastLED/blob/master/Arduino/LEDstream_FastLED/LEDstream_FastLED.ino I have no issues with boards crashing when receiving data from LedFX, but no luck when running WLED.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-04-04:
Thank you for reporting this!
How many LEDs are you sending via Adalight?

apophisnow, 2022-04-04:
I was testing with 22, but I don't think the count is part of the issue
necessarily.

Bene2103, 2022-04-06:
I experienced this behaviour too. But i downgraded to 0.13.0b6.

I also use Adalight protocol but with boblight. In my setup i have 220 LED's.

Thanks for your support!

sh00bx, 2022-06-26:
I've created a hacky workaround for this issue on latest, making these changes:
- wled_serial.cpp dumbed down to basic adalight functionality only (removed serial out, baudrate customization, improv and tpm2)
- FastLED.show() instead of strip.show() in wled_serial.cpp
- hardcoded adalight serial baudrate @921600 in wled.cpp / wled_serial.cpp

See [sh00bx-adalight-workaround](https://github.com/sh00bx/WLED/tree/sh00bx-adalight-workaround)

No more crashes with these modifications on 0.13.2a0 but to my surprise this did not also resolve another issue I have with adalight on serial, glitches in conjunction with wifi. But I'll open a separate issue on that.

blazoncek, 2023-09-28:
If the issue persists in 0.14.0-b6 or later please reopen the issue.

closed 2023-09-28 by blazoncek

AZDane, 2023-10-01:
I see this behavior in 14.0b6. Using ESP32.

d3d9, 2024-06-03:
Can this be reopened as it still occurred on v0.14.0-b6 in October and for me just now on v0.14.4?

I am using an ESP 8266 with 32 pixels and previously used DDP over wifi from LedFx. Now I wanted to try adalight/serial.  
This stops working after 1 or 2 seconds. Using one of the diagnostic LedFx effects like Metro or Pixels I could avoid the crash, but switching to something that lights up more pixels in different ways causes the crash almost instantly.  
Though I'm not sure if crash is the right term. The WLED uptime doesn't get reset. It just doesn't react to further serial input at all until WLED is restarted. Which further details can I provide?