wled/WLED #2623: Latitude: max 66.6 ?

Quato38, 2022-04-08:
### What happened?

I live here: Latitude: 69.64, Longitude: 18.99  so i can't get exsact location on my wleds

When i enter 69.64 in Location settings i get an error:  Latitude: must be less og equal to  66.6

### To Reproduce Bug

enter Latidue: over 66.6 

### Expected Behavior

i expected to enter 69.64 witch is my latidude

### Install Method

Binary from WLED.me

### What version of WLED?

wled 13.01

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-04-08:
Unfortunately you live above polar circle where sunrise and sunset calculations will not work.

closed 2022-04-16 by blazoncek

Aircoookie, 2022-04-16:
Added this info to KB at https://kno.wled.ge/features/presets/#applying-presets-at-sunrise-and-sunset