wled/WLED #2624: "Failed updating firmware!" 0.13.0-b6 to 0.13.1

LeducH, 2022-04-08:
### What happened?

Updating to 0.13.1  returns 
"Failed updating firmware!"

### To Reproduce Bug

Install [0.13.0-b6] in a 8266

try to update to  0.13.1 wit h released bin file 

### Expected Behavior

Successfully updated to 0.13.1

### Install Method

Binary from WLED.me

### What version of WLED?

Server message: WLED 0.13.0-b6 (build 2112080)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-04-08:
Use web flasher if OTA fails.

Aircoookie, 2022-04-08:
Likely you have used the wrong flash size for your board. What type of ESP8266 board is it in particular, and what exact binary did you try?

LeducH, 2022-04-09:
> Likely you have used the wrong flash size for your board. What type of ESP8266 board is it in particular, and what exact binary did you try?

Is used [WLED_0.13.1_ESP8266.bin](https://github.com/Aircoookie/WLED/releases/download/v0.13.1/WLED_0.13.1_ESP8266.bin)

and my board is the  this one

sophof, 2022-04-09:
I tried updating from b07 to 0.13.1 and it kept failing as well. USed the same binary to OTA manually and the device became unresponsive (no log even when connecting to USB). Had to wipe everything on a reinstall :(

toto79, 2022-04-12:
@LeducH : Please give 

@LeducH: Please try WLED...ESP02.bin ... it is build for all ESp8266/8285 with 2MB Flash Size. All of my Athom Bulbs have 2MB Memory

LeducH, 2022-04-13:
> @LeducH: Please try WLED...ESP02.bin . It is build for all ESP8266/8285 with 2MB Flash Size. All of my Athom Bulbs have 2MB Memory

@toto79 : thanks I will give a try. I will update this post how it goes.

closed 2022-04-13 by LeducH

f1ren, 2023-07-17:
For `v0.14.0-b3` see [this answer](https://github.com/Aircoookie/WLED/issues/3257#issuecomment-1607091340).