wled/WLED #2627: Upgraded to 0.13.1 and IR does not work anymore

rb0135, 2022-04-11:
Hi,

Just checking if IR has been broken.

Previous verions, I have a lamp running with WLED and a 24 key cheap remote. This has been working for a while now. The WLED unit got updated to 0.13.1 (through HomeAssistant) and now the IR Remote does not work.

I did search, but cant find anything specific.

I have reapplied all settings, but not working (even replaced the battery in the remote).

So, that begs the question if anyone else has had the issue of IR remote stop working with 0.13.1

Thanks,
Rob

blazoncek, 2022-04-11:
IR is working in 0.13.1.
Try manually updating (OTA) by downloading binary from Github.

rb0135, 2022-04-11:
Thanks for the suggestion.

I tried your suggestion and it still does not work. IR has definetly been broken in my case. Would the codes have changed for the 24 Key RGB remote? I have 6 of these remotese and when the IR was working, only 1 of these remotes actually worked (even though they looked identical). I have tested the 6 remotes and none work now.

blazoncek, 2022-04-11:
IR has been updated to support custom IR codes without recompiling source code but existing codes were not changed.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
You may as well look up https://kno.wled.ge/interfaces/infrared/
Closing.