wled/WLED #2632: Flashing of ESP8266 07 / 12  over Web Instaler does end with "Timed out waiting for packet header"

mschlen, 2022-04-13:
Hi together,

recently i got in touch with this amazing project. For an own project I need every space i can get, so I want to use the WLED purely on an ESP8266 07 or 12 without additonal components. For that I tried to flash the chip via the Web Installer, what should be very easy I think. Now I have the problem, that flashing goes very well, but in the end it says "Timed out waiting for packet header". If I connect the ESP than to normal running, no Wifi AccessPoint comes up to enter settings. I attached a file of how I have wired the ESP while flashing.  I am sitting quite a few days also searching on the internet for this error notice, but I can not solve it. Hope somebody of you can give me a hint what I am doing wrong in this setup.

Thanks very much in advance
Markus

[image]

blazoncek, 2022-04-13:
Please use WLED [forum](https://wled.discourse.group) or Discord for help or support questions.
Other than that you need to select proper binary for your ESP (flash size matters). Check your ESP flash size if it is 2MB use ESP02 binary if it is 1MB use ESP01 binary (ESP01 may not be fully supported using web flasher ATM).
Closing.

closed 2022-04-13 by blazoncek

mschlen, 2022-04-13:
ok, thanks. I will try you hints and if it does not work ask in forum or Discord.