wled/WLED #2633: LED turns power on after reboot/power up

guergen1, 2022-04-14:
### What happened?

The LED turns on after reboot or powered up, the switch ist not on:
[image]

### To Reproduce Bug

After downgrade to 0.13 b 1 it works fine

### Expected Behavior

The LED do not turn on!

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 (build 2203150)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-04-14:
Thanks for the report.
Do you have a boot preset configured?

guergen1, 2022-04-14:
No, there is no boot preset configured:
[image]

Cryptophunk, 2022-05-03:
I've had this issue too. I've solved it temporary by setting a reboot schedule everyday using an external plug.

modev92, 2022-05-06:
I have this issue too, but defined one preset to have a different starting color. But the toggle: Turn Led on after power up/reset is off.

RestOp, 2022-07-10:
I have other begaviuor - i do NOT want to turn led on after power loss. But it turns on inspite of it!

heyitsyang, 2022-11-22:
I see the same thing but it happens when I restart Home Assistant also.  I have Turn on LEDs after boot/reset unchecked and I have a boot preset set.  If I restart Home Assistant, WLED turns on to the preset.

R1DEN, 2023-01-31:
have the same issue as @heyitsyang

blazoncek, 2023-01-31:
As a workaround create a preset with API `{"on":false}` and assign it as a boot preset.

closed 2023-01-31 by blazoncek

heyitsyang, 2023-01-31:
I appreciate the tip about the workaround, but a workaround is not a fix.  Shouldn't this issue remain open?

blazoncek, 2023-01-31:
As a matter of fact I am unable to reproduce this kind of behaviour on 20+ of my devices. With or without boot presets or on/off toggle setting.
As such I deem problem not WLED related.

RestOp, 2023-01-31:
I had v11, now updated to last ( 0.13.3) - now works as expected! 
[image]
After disconettiong of power it blikns one time and then stays in OFF state.

Thanks a lot!

marcolino7, 2023-10-15:
Hi,
I have 4 node on version 0.14
One of this node do not power on leds on reboot, other yes. Same config, same boot preset on all 4 nodes.
Option to turn On led on after power up is unchecked. Something else I could verify?

Thanks

jabber77, 2023-11-16:
Hi, 
I have the same problem except that the problem appeared when I added the led strip to Home Assistant.
WLED version 0.13.3

jabber77, 2023-11-16:
Hi, 
I have the same problem except that the problem appeared when I added the led strip to Home Assistant
WLED version 0.13.3

marcolino7, 2023-11-17:
> Hi, I have the same problem except that the problem appeared when I added the led strip to Home Assistant. WLED version 0.13.3

Hi,
I use this startup preset and I fixed:

`{"on":false,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[127,172,255,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

raspberrycoulis, 2024-04-25:
I have this issue, running 13.3 as anything newer seems to mess up WiFi connectivity. 

Basically, fresh install on an ESP8266 with 100 LED's connected, the turn LED's on by default is unchecked and preset is set to 0. However a reboot causes the LED's to power up still!

Have Home Assistant configured too, so tried removing the node from there in case it was linked (as others have mentioned) but still have the issue. 

Any more ideas as to why this is happening?

blazoncek, 2024-04-25:
The only change (if needed) will happen in 0.15 or newer so you will need to upgrade unfortunately.

raspberrycoulis, 2024-04-25:
> The only change (if needed) will happen in 0.15 or newer so you will need to upgrade unfortunately.

Hmm ok. Since 0.14 my devices all become unresponsive. On the off chance the UI does partially load, there are browser console errors with things failing to load or the device loses connection to the LED's.

I may test a beta 0.15 and see what happens.

DedeHai, 2024-10-05:
> The option "Turn LEDs on after power up/reset" shoud be independant of the preset.

I disagree. Apply preset at boot means that it applies that preset at boot. If you want that preset not to turn on the LEDs, do as you proposed (on:false)

markusressel, 2024-12-13:
I want to change the brightness and color using a preset, but not affect the "on"/"off" state, because a PIR Sensor is controlling this. However, even when removing the "n´on" state, as you suggest, the light still turns on.

I cannot remove the "brightness" property because I need to control that even when the light is off. Is there a way to do this?

blazoncek, 2024-12-13:
@markusressel see [KB](https://kno.wled.ge/interfaces/json-api/).

You are not doing it right. Anything, including what you request, is possible by correctly utilising JSON API.

markusressel, 2024-12-13:
@blazoncek I don't have access to the API because I only have the device itself. But I guess I am " not doing it right" by wanting to have a device that works on its own. Thanks anyway for the useless response 👌

For anyone interested: This is easily possible using ESPHome and that's what I am have done now. WLED is fine for simple usage (and API usage apparently), but on device seems to be lacking. Love to both of them, cheers.

blazoncek, 2024-12-13:
Apparently you do not want to learn. Sorry.

BTW anyone has access to JSON API. In fact everything is driven by JSON API including UI and external and internal actions. Just look into preset - that's your JSON API.