wled/WLED #2634: wireguard

fkoteam, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
My home assistant is on cloud and I need a wireguard connection to connect to my devices.

**Describe the solution you'd like**
It would be great if wled could be support wireguard.

**Describe alternatives you've considered**
The only alternative is try to program it by my own. Is there a reason that could makes it non-viable?

Thank you for your ideas for making WLED better!

blazoncek, 2022-04-15:
Support in what way?
As I understand Wireguard is a VPN solution and as such connects devices from WAN to (W)LAN. WLED normally resides in WLAN.
In such case your router should support Wireguard, not WLED.

fkoteam, 2022-04-15:
Esp32 and esp8266 can support wireguard directly. I have a esp32cam connected to my home assistant cloud in this way.
Sadly, I can't change the firmware of my router and I'm behind of a cgnat.

blazoncek, 2022-04-15:
Since you have an incentive I encourage you to implement it and share it by submitting a PR.
According to the provided example it should be more or less straightforward.

If you need assistance join Discord.

Note: I do not have neither the will nor means to implement it myself.

BTW cheap EdgeRouter X from Ubiqiti can support Wireguard.

blazoncek, 2023-08-05:
Will be solved by #3270

closed 2026-02-07 by softhack007