wled/WLED #2639: USERMOD_AUTO_SAVE feature not working

asking23, 2022-04-19:
### What happened?

To Enable Last State Save option, recompiled using PlateformIO.

Got below option in Web UI. But even after settings its not saving last state.

Added to PlateformIO.ini File
` -D USERMOD_AUTO_SAVE feature`

[image]

### To Reproduce Bug

Just turn off ESP32 after 15 Seconds and rebooted but no effect. Same Default light setting is picked up. 

### Expected Behavior

Expecting to store the last state when power fails.

### Install Method

Self-Compiled

### What version of WLED?

WLED-0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-04-19:
Unfortunately I cannot confirm your findings.
Auto Save usermod is working as expected.

asking23, 2022-04-20:
You're right. Its working as expected. 

Just those who are looking for Saving last state Please remember to uncheck Powerup/reset and Make Presets : 0.

[image]

closed 2022-04-20 by asking23

blazoncek, 2022-04-20:
Will add this check to Auto Save usermod...