wled/WLED #2641: Magic Home ERROR-10

neuherzk, 2022-04-22:
### What happened?

Hello Gents,

I as able to flash my MagicHome LED-Controller with WLED 0.13.1 ESP8266.bin but I get error 10, and after every reboot it switch back to WLED-AP.
I have dont the flash like the guide here:
https://github.com/Aircoookie/WLED/issues/398

### To Reproduce Bug

Flash the controller, setup my Wifi, reboot device.

### Expected Behavior

I would expect that the ESP will safe the config.

### Install Method

Binary from WLED.me

### What version of WLED?

0.31.1 ESP8266.bin

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-04-22:
You should use correct binary.
Check the flash size in your controller and select proper binary.

tiigaaa, 2022-04-22:
Hi,

thanks for replay, but in the closed issue they used the ESP8266.bin is this the wrong for ESP8285, I dont have seen a special version...

blazoncek, 2022-04-22:
8266 but with correct flash size.

tiigaaa, 2022-04-22:
Ahh, yes
With tha verion it is working. "WLED_0.12.0_ESP8266_1M.bin"

:)

closed 2022-04-22 by blazoncek