wled/WLED #2643: using HTTP json with ps:1 result in power on even with on:false

nodiaque, 2022-04-25:
Hello,

I am using http json to set this:

`"{"on":false,"bri":37,"transition":7,"ps":1,"pl":-1,"nl":{"on":false,"dur":60,"mode":1,"tbri":0,"rem":-1},"udpn":{"send":false,"recv":false},"lor":0,"mainseg":0,"seg":[{"id":0,"start":0,"stop":300,"len":300,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[0,72,255],[255,255,255],[68,255,0]],"fx":93,"sx":73,"ix":233,"pal":0,"sel":true,"rev":false,"mi":false}]}"`

In short, this put an animation, set brigthness and turn it off. I've updated to 13.1 and now when I send this, I see the led strip kind of shutting down but turning back up. I played with the query an as soon as I remove the ps:1, it stop the problem.

Hick is I use nodered for some automation and what I do is query the json actually in place and I save it, and after a certiain event I restore this saved json. So it include everything. I don't know what ps does, but it's screwing with the power right now.

thank you

blazoncek, 2022-04-26:
`"ps":1` loads preset 1 which overrides all other options.

nodiaque, 2022-04-28:
Ok, but this is what the json query send me. The exact string on top is what I receive, and since I copy/paste that string, I do not parse anything.

blazoncek, 2022-04-29:
Please read the documentation at: https://kno.wled.ge

What WLED sends as a response to ‘/json‘ query is a **state** so if you restre that you restore WLED state which includes running preset.
As specified above, preset may override any state value.

Closing as this is user error.