wled/WLED #2649: ESP32 DevKit resets if you try set up more than one LED output via the Android app

JackiePease, 2022-04-30:
### What happened?

On selecting Save from "LED and Hardware setup" after setting up more than one LED output, the ESP32 resets to original settings

### To Reproduce Bug

Connect via Wifi using the Android App.
In LED preferences, change Length on output 1 to 300 and Save (you can close and reopen the app to check it has saved) 
In LED preferences, add a new LED output. Set Length to 300 and GPIO to 12 then Save
Go back into LED preferences
Only one LED output is set up, it has length 30 (all other settings are also lost)

### Expected Behavior

You should be able to setup more than 1 LED output
I am able to get round this bug by formatting the LED outputs via WLED.me

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.1 and 0.13.0-b6 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Changchung, 2022-05-02:
I am having the same issue. Is very frustrating. And in my case even the RGB colors wont match even being using the same Leds on each output.

Aircoookie, 2022-05-03:
Thank you for the report! Does the issue also occur if you attempt to use GPIO 1 or 3 for the second output?

Changchung, 2022-05-04:
> Thank you for the report! Does the issue also occur if you attempt to use GPIO 1 or 3 for the second output?

Thanks Aircookie. Yes. I being trying different boards and different outputs like, 16, 1, 3, 26, 12, 13, 14, 15. It lost everything. All the Led settings. I try with the 12 versions and seeing be ok. Problems is with the 13. More people on the chat forum are having the same issue. And another thing that I noticed is that when you have all the outputs working and having same leds on the outputs they wont match in color amber and the first output is bright and the rest dim. When you start changing colors and brightest and patterns they match. Here is a video about how the 13 erase the settings. Or it wont saves them. Or crashed.

https://youtu.be/RUUjeQGJE-U

JackiePease, 2022-05-04:
> Thank you for the report! Does the issue also occur if you attempt to use GPIO 1 or 3 for the second output

Sometimes - e.g. 
Set  output 1 to GPIO 2 Length 300 
Set output 2 to GPIO 3 Length 300
Save
OK
Change output 2 to Length 325
Save 
Resets

But this doesn't seem to be consistent.

Changing output 2 GPIO from 3 to 13 seems to cause a reset.

Get back to me if you need me to try something else.

Changchung, 2022-05-05:
I just realized that it erase the presets if changes the RGB order. 
Set output 1 GPIO 14 1 length 20
Save go to effects select any.
Saved like a preset
Go to Led settings to change RGB order.
Save, presets gone.

iStarLights, 2022-06-08:
> Sometimes - e.g. Set output 1 to GPIO 2 Length 300 Set output 2 to GPIO 3 Length 300 Save OK Change output 2 to Length 325 Save Resets
>
> But this doesn't seem to be consistent.
>
> Changing output 2 [...]

Ive been having the exact same troubles as well on 20 different devices (different networks.)

Ive notices if I let it reset, then reboot the system, then enter in all the led outputs correct the first time and leave it, that seems to work.

nwalke, 2022-06-27:
I am running in to seemingly this issue as well, but it's not limited to the android app.  I cannot successfully configure a second group of LEDs using the latest version of WLED, but 12.0 works just fine.

This works in 12.0:
[image]

This works in 13.1:
[image]

This does NOT work in 13.1:
[image]

Again, any single strip configuration seems to work just fine but I cannot get them both to work.  I'm seeing it occasionally reset when I go through the Android app, but not the web app.  It just doesn't work when I configure it through the web interface.  When I did an OTA update back to 12 it still would not work until I deleted both configs and readded.

Cryptophunk, 2022-06-27:
I too have had this problem. I've had this problem with pre flashed esp32s version on 12, and 13.1. Rebooting seems to allow me to set my setting but I have to get my options right the first time.

closed 2024-04-16 by github-actions[bot]