wled/WLED #2650: Touch Button is not working

masx14, 2022-04-30:
In the 0.13.1 release the touch button is not working. 

The buildflag to enable it is still present in the _platformio_override.ini.sample_ file.

I have looked into the Code and realized that the problem is in the _cfg.cpp_ file. 
The buildflag TOUCHPIN is ignored and therefore the pin is not set in the btnpin array. 

By setting the btnpin array manually I have managed to get it working, but it is more like a quick & dirty hack.

It would be great if someone could look further into that, because I don't understand the code well enough.

blazoncek, 2022-04-30:
There is no more need to set build flags to use touch buttons on ESP32.
Just select appropriate GPIO (touch enabled) in the hardware settings (LED Preferences) page and button type of Touch.

closed 2022-04-30 by blazoncek

masx14, 2022-04-30:
I did not expect that. 

thank you