wled/WLED #2651: Sunrise Time Is not Correct

Muditguptagolden, 2022-05-02:
### What happened?

Sunrise Time is not calculated it shows 00:00 where as the sunset time is correct.
And the Time is also correct. 
Even though I have used the compiled the latest version from PIO.
I have WLED version 0.13.2-a0.
And micro-controller ESP32.
I have uploaded the screenshot for the same.
[image]

### To Reproduce Bug

Install WLED in ESP32
Open Settings>Time & Macros> 
Set NTP time.google.com>Time Zone IST(INDIA) & UTC Offset 0>Latitude And Longitude to 26.53'N & 80.48'E> 
Hit Save  

### Expected Behavior

It should have shown Sunrise 5:30 and Sunset 18:41

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.2-a0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Cryptophunk, 2022-05-13:
Try using 0.pool.ntp.org and rebooting. That did the trick for me.

blazoncek, 2022-09-21:
There are (rare) edge cases where our custom sin_t/cos_t functions (wled_math) will produce incorrect results. These errors should go away within a day or two (one or two missing sunrise/sunset triggers).
You can alleviate the problem by self compiling and adding `-D WLED_USE_REAL_MATH' to the PIO environment. Be warned that this will use quite a few more bytes of RAM on ESP8266.

closed 2022-09-21 by blazoncek

blazoncek, 2023-05-01:
WORKAROUND: if you compile with PIO and add *build_flag*`-D WLED_USE_REAL_MATH` then WLED will use standard *libc* functions instead