wled/WLED #2654: MQTT not reconnecting on network loss

nodiaque, 2022-05-03:
### What happened?

Home network failed. Wireless was still up but DNS down, thus no dns routing. At the sametime, mqtt server died. Once everything was back up (both docker), all WLED device were non responsive even after many hours on mqtt. I had to connect directly to each of them and issue a device restart for them to reconnect.

### To Reproduce Bug

Connect to an mqtt server wich user/pass. Shutdown inexpectly the mqtt server (not a gracefull with a shutdown message) and maybe dns too. Wait maybe 2 hours (can't say how much time to wait since both time I had a server crash, it lasted more then 2 hours since I was away). Spin-up mqtt and dns.

### Expected Behavior

WLED should reconnect to the mqtt server

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-05-04:
Please make sure your case is 100% reproducible. Repeat the event and if you can 100% reproduce it I will provide you with debug build so you can monitor what is happening on ESP at the time of event.
It is difficult (or almost impossible) to diagnose one-time events without any logs available.

nodiaque, 2022-05-28:
I can reproduce it quite easily, just happened right now. Simply reboot your router and let them reconnect to the network, and bam, no more mqtt. I currently have 2 wled not working through mqtt right now. What do you want me to do?

blazoncek, 2022-05-28:
Please use a debug build from [here](https://github.com/srg74/WLED-wemos-shield/tree/master/resources/experimental/Firmware/Debug_builds) and upload it to one of ESPs (OTA or serially). If you can you can also compile yourself, just add -D WLED_DEBUG to build flags.

Then connect it to computer using USB cable and monitor serial output (any program that can read COM port is ok, but most use Arduino IDE or MS VSC + PIO). You don't need to have LEDs connected.
Pay attention to the output when network disconnects and when it reconnects again. Post results here.

blazoncek, 2022-07-07:
Since there are no updates I assume the problem went away.
If the problem persists, please update the issue otherwise I'm going to close it.

softhack007, 2023-12-30:
Stale since 18 months, with no response from original problem author.

Closing.

closed 2023-12-30 by softhack007