wled/WLED #2655: Adding more than one button

romanito123, 2022-05-03:
Hi there, I am trying to connect two buttons to my ESP 8266 to trigger presets. I connected the first button to D3 and left Button 0 on 0 and it works perfectly. However no matter what GPIO port i plug the second button into I never get a response and anything above d4 shorts the board.  Going into the LED preferences and changing the number also has no effect. I feel like it is a very simple thing that I am doing wrong but cant seem to find a video or forum of anybody connecting more than one button. The picture below shows how i have it set. So yeah button zero is definetly the onboard flash button/D3 as well but cant get anything else to work and moving button 0 off anything but 0 stops it from working.
Would greatly appreciate some help on this as only just updated my boards so only just discovered this feature

[image]

blazoncek, 2022-05-04:
Please provide a description or close the issue.

Aircoookie, 2022-05-04:
Original description was the following, thus i assume they got it to work:

"Hi there, I am trying to connect two buttons to my ESP 8266 to trigger presets. I connected the first button to D3 and left Button 0 on 0 and it works perfectly. However no matter what GPIO port i plug the second button into I never get a response and anything above d4 shorts the board. Going into the LED preferences and changing the number also has no effect. I feel like it is a very simple thing that I am doing wrong but cant seem to find a video or forum of anybody connecting more than one button. The picture below shows how i have it set. So yeah button zero is definetly the onboard flash button/D3 as well but cant get anything else to work and moving button 0 off anything but 0 stops it from working."

closed 2022-05-04 by Aircoookie