wled/WLED #2656: Mqtt -> qos and retain not set right

KrX3D, 2022-05-04:
### What happened?

hello,

im trying to creat an usermod which send out some mqtt parameters, but the qos and retain is not set right.

doing something like this:

```
char buffer[64];
sprintf_P(buffer, PSTR("%s/config/seg/%d/Red"), mqttDeviceTopic, CurrSeg);
mqtt->publish(buffer, 2, true, String(ColR).c_str());	
```

and i receive in Home Assistant -> QoS: 0 - Retain: false

i searched a bit and found in mqtt.cpp from line 124 this:

```
char s[10];
char subuf[38];

sprintf_P(s, PSTR("%u"), bri);
strlcpy(subuf, mqttDeviceTopic, 33);
strcat_P(subuf, PSTR("/g"));
mqtt->publish(subuf, 0, true, s);         // retain message
```

which should  get a result like this: QoS: 0 - Retain: true but it also is received as QoS: 0 - Retain: false

[image]

### To Reproduce Bug

```

		char buffer[64];
		sprintf_P(buffer, PSTR("%s/config/seg/%d/Red"), mqttDeviceTopic, CurrSeg);
		mqtt->publish(buffer, 2, true, String(ColR).c_str());	
```

on home assistant, mqtt broker receives: QoS: 0 - Retain: false

### Expected Behavior

QoS: 2 - Retain: true

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-05-05:
Have you checked other MQTT tools? I am seeing correctly retained messages.

KrX3D, 2022-05-05:
Ok thanks for the info, i tried Mqtt Explorer and there it shows that is it retained, but the qos is still 0  ?

[image]

blazoncek, 2022-05-05:
Check with AsyncMqttClient library about QoS since WLED uses that.

Aircoookie, 2022-09-23:
Just checked on this, we are using an anchient version of AsyncMqttClient, which did not support all higher QoS levels.
You can try the `newmqtt` branch I just made, which updates the library. Would like to avoid updating the library on the main branch unless it fixes a bug, since updating it increases flash usage slightly.

As a side note, I would choose QoS level 1 instead of 2, 2 only protects against the same message arriving multiple times, to ensure message is delivered QoS 1 is plenty.

closed 2022-09-23 by Aircoookie