wled/WLED #2660: Aplication hangs after 24hours on NodeMcu V3

Hubert97, 2022-05-06:
### What happened?

Module runs normally. After aproximately 24 hours it becomes unresponsive - it is not present on network. Restarting board fixes the problem.

### To Reproduce Bug

Node mcu V3. after approx 24 hours module goes offline.

### Expected Behavior

For it to stay functional like in prevous version.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Prevous version i used on this module (10 something) worked fine for a long long time without probblems. After upgrading it behaves like this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Hubert97, 2022-05-07:
Update on bug. It seems that the module loses wifi connection. I turned animation on after about 15 hrs from initial power up, and it worked fine. However after ~24 hrs it is no longer present on my network tho animation is still present. I checked and it also doesn't create its own ap.

poelzi, 2022-05-08:
Does the button work to toggle master on/off. It could be a hard hang which I experience a lot with current FW. I added added a watchdog so the node does not hard lockup anymore #2657

Hubert97, 2022-05-09:
I do not have an easy access to button and also i don't use it. Ill checkout your commit and reply if this solved the issue.

Phara0h, 2022-08-30:
I have the same issue and the toggle button still works even though it's not reachable via wifi

Hubert97, 2022-09-28:
Problem is still persistent on recent build.

Hubert97, 2022-10-06:
I've noticed that when aplication hangs it is still discoverable by other machines - responds to ping.

blazoncek, 2022-11-18:
FYI similar behaviour can be observed on ESP32-S2, effects still play, buttons still work but device is unreachable over WiFi.
This is not a WLED bug *per se* but rather underlying WiFi/AsyncWebServer implementation bug.
I am closing this issue as WLED can do little to mitigate the problem. Newer versions of libraries may improve the behaviour in the future.

closed 2022-11-18 by blazoncek