wled/WLED #2663: Two values for maximum current for RGBW

Markkuuss, 2022-05-08:
I use the feature "automatic brightness limiter" with a SK6812 LED strip.

Unfortunately I found out that the value "custom max. current per LED" fits either for the RGB part of the LED or for the W part.

From my point of view it would be useful to have two separate values for the max. current for a RGBW LED. Once the max. current for RGB and once for the white LED.

blazoncek, 2022-05-09:
How would you handle a situation when all LEDs are lit? They are drawing current from the same source.

Markkuuss, 2022-05-09:
That's a good question. Which algorithm is the "automatic brightness limiter" feature based on? I assume that the power consumption is calculated based on the currently set color. If the current consumption exceeds the maximum specified current, the brightness is reduced?

I have the impression that the calculation for automatic brightness limiter does not work correctly for SK6812, because possibly the wrong values for the current consumption of the W channel are defined.

It would be most useful if you could set the maximum current for each channel - i.e. four setting fields for max. currents and not just one.

closed 2022-11-01 by blazoncek